FreeCAD/FreeCAD #20929: Sketch: Invalid input error on windows

adrianinsaval, 2025-04-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

sketcher appears to be completely broken on the latests windows weekly and my own pixi build, seems to be a problem with the toponaming stuff. Adding geometry to existing sketches breaks them and on new sketches only the first geometries added on a sinlge command work, any further additions break the sketch.

reproduction steps:

new file
new sketch
add a line and exit sketch => works
edit sketch and add a second line => sketch is broken, report view says:
```
19:55:21  <Exception> Invalid input
19:55:21  Sketch: Invalid input
```

this works with any geometry not just lines, but is not simply that any second geometry breaks the sketch, multiple lines can be added without issues if they are created by the same action, for example a single rectangle or a polyline with multiple lines don-t trigger the issue unless you add a second set of lines

at some point I also got the error `Illegal character in mapped name` but I have not been able to reproduce that one

I was not able to reproduce on the appimage

### Full version info

```shell
OS: Windows 11 build 22621
Architecture: x86_64
Version: 1.1.0dev.41354 (Git)
Build date: 2025/04/21 18:01:03
Build type: Release
Branch: main
Hash: fd390acd53dae91898145a22d74050d0d648a45b
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

adrianinsaval, 2025-04-22:
I tried now a libpack build and it looks like it works correctly so it seems to be a pixi specific bug

adrianinsaval, 2025-04-23:
to clarify, this bug is present on the new weekly build from https://github.com/FreeCAD/FreeCAD/releases/tag/weekly-2025.04.21 I built one at https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds and that one does not have the bug

maxwxyz, 2025-04-23:
@AjinkyaDahale @oursland FYI

AjinkyaDahale, 2025-04-27:
Is this a crash or things just not working? If the former can we get a backtrace?

AjinkyaDahale, 2025-05-04:
Is this still happening? One more trim PR (I forget the number) was merged since this issue was reported.

maxwxyz, 2025-05-13:
On the latest weekly in the main repo, there are error messages as soon as geometry is added ^

[image]

Maybe also related to #21207 ?

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/05/13 09:39:01
Build type: Release
Branch: main
Hash: 7d57569eb493b929e5647bd9b0ad03b33d1acc45
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/133.772/2
Installed mods: 
  * Cables 0.1.4
  * CfdOF 1.30.1
  * CurvedShapes 1.0.13
  * Curves 0.6.61
  * dodo 1.0.1
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * Gridfinity 0.11.7
  * Manipulator 1.5.9
  * OpticsWorkbench 1.2.0
  * parts_library
  * Rocket 4.1.1
  * sheetmetal 0.7.22
  * Silk 0.1.5
  * Telemetry 1.0.0

```

adrianinsaval, 2025-05-14:
> Maybe also related to [#21207](https://github.com/FreeCAD/FreeCAD/issues/21207) ?

it's the same bug IMO

adrianinsaval, 2025-05-14:
#20961 is also a duplicate

Budman1758, 2025-05-15:
Build 41585 on Windows.  Exact same thing.  New file, new sketch, draw and dimension a rectangle no problem. Draw a second rectangle and instant "Invalid input" and sketch is broken.

karamellpelle, 2025-06-11:
> to clarify, this bug is present on the new weekly build from https://github.com/FreeCAD/FreeCAD/releases/tag/weekly-2025.04.21 I built one at https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds and that one does not have the bug

I found this issue after testing the newest weekly _FreeCAD/FreeCAD/releases_. _FreeCAD/FreeCAD-Bundle/releases_ works.

## Weekly FreeCAD/FreeCAD-Bundle/releases (working)

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.42133 (Git)
Build date: 2025/06/09 17:51:59
Build type: Release
Branch: main
Hash: 12b4610b499a9756f19bbb85f59fe16e5fa135db
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Turntable/Window center
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.047/1.5
Installed mods: 
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * IconThemes
  * OpenTheme 2025.5.20
  * sheetmetal 0.7.22
```

## Weekly FreeCAD/FreeCAD/releases (failing)
```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/06/09 17:51:59
Build type: Release
Branch: main
Hash: 12b4610b499a9756f19bbb85f59fe16e5fa135db
Python 3.11.13, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Turntable/Window center
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.047/1.5
Installed mods: 
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * IconThemes
  * OpenTheme 2025.5.20
  * sheetmetal 0.7.22
```

I didn't know there were two different kinds of releases, each in their own repo.

maxwxyz, 2025-07-27:
I cannot reproduce with  https://github.com/FreeCAD/FreeCAD/releases/weekly-2025.07.26/ 
If someone encounters this, please comment!

closed 2025-07-27 by maxwxyz

karamellpelle, 2025-07-27:
* The newest (_weekly-2025.07.21_) [FreeCAD/FreeCAD](https://github.com/FreeCAD/FreeCAD/releases/tag/weekly-2025.07.21) development build does not have a Windows variant now.
* The newest (_weekly-builds-42721-conda-Windows-x86_64-py311_) [FreeCAD/FreeCAD-Bundle](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)  development build seems to work: I created a simple Sketch and padded it without error.
* Even more interesting: I was finally able to sucessfully build and run (padding a simple Sketch without error) the newest commit (91641c4fb6 / 42736) on Windows using _Pixi_ 😃 . 

Good progress has happened - thanks!

maxwxyz, 2025-07-27:
@karamellpelle check out  https://github.com/FreeCAD/FreeCAD/releases/weekly-2025.07.26/

karamellpelle, 2025-07-27:
> [@karamellpelle](https://github.com/karamellpelle) check out https://github.com/FreeCAD/FreeCAD/releases/weekly-2025.07.26/

Sorry, I was looking for the newest but didn't see that. For what I understand, the/a difference between the development builds at **FreeCAD/FreeCAD** vs **FreeCAD/FreeCAD-Bundle** is that the first one uses _pixi_ and the other _FreeCAD-LibPack_. Hopefully this means that the pixi build has been fixed, as I just experienced  :)

oursland, 2025-07-27:
The second does not use FreeCAD-LibPack, but is a different approach of using `conda-forge` which involves 3 repositories.  The problem with the `FreeCAD-Bundle` is that weekly builds are not persisted and tracked, and there are no strong links between a given weekly build and source code.

The `FreeCAD/FreeACAD` weekly builds are tagged releases which means older builds can still be downloaded, which is beneficial for identifying when a bug was introduced.  As these builds are tagged, we can employ tooling such as `git bisect` to narrow down to the individual commit that introduced a bug.