FreeCAD/FreeCAD #20930: [BIM/DRAFT] Correct the activation of axes using keys (language layout) and with the shift key pressed.

kaiwas, 2025-04-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

At the moment, when creating an object, the XYZ keys do not work if the keyboard layout is different from English.
When pressing XYZ, the focus is not on the desired axis.

We can take into account the direction only with the shift key pressed. But in this case, the NumPad does not work for entering values.

This applies to both creating objects and the move/scale/rotate tools and all others where there are XYZ fields.

Video demonstration.

https://github.com/user-attachments/assets/dc2b4aca-cffe-4564-a4ff-1d6cc9969725

You can see on the screen pressing **Ч**. This is Х on the Russian layout.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41354 (Git) AppImage
Build date: 2025/04/21 18:01:03
Build type: Release
Branch: main
Hash: fd390acd53dae91898145a22d74050d0d648a45b
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1
Installed mods: 
  * Reinforcement
  * SvgWorkbench-main 1.0.0.dev5
  * PitchedRoof
  * Road 2025.2.22
  * PitchedRoof.zip
  * TrackingGroups
  * Curves 0.6.61
  * freecad.gears 1.3.0
  * TrackingGroups.zip
  * sheetmetal 0.7.22
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

BIM

### Anything else?

Addition. To set the object to 0/0/0, I have to move the mouse along all the panels so as not to accidentally get to the 3D window. This is terrible!

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-04-23:
I think this might be an issue on non-[Latin-script keyboards](https://en.wikipedia.org/wiki/List_of_Latin-script_keyboard_layouts). I have a German keyboard and pressing either X, Y, or Z to restrict movement on the associated axis works.

I presume you have a Cyrillic script keyboard layout, and from the version info, you're using the matching `ru_RU` locale. I don't understand why this does not work. Perhaps it is an issue with untranslated keyboard shortcuts?

kaiwas, 2025-04-23:
> Perhaps it is an issue with untranslated keyboard shortcuts?

Yes. I have a Cyrillic layout.
I don't know what exactly the problem is. Perhaps the display environment (KDE) can also affect it.
I can't check how 1.0 works on Windows (this version doesn't start on Win7). But I'll check later how 0.21 behaves

semhustej, 2025-04-23:
> 
> Can you check how the combination with SHIFT+Numpad works? Start building a wall. Hold down SHIFT and set the direction with the mouse cursor. Without releasing SHIFT, press the numbers on the numpad. Will the fields accept keystrokes?
This does not work. You have to release the Shift key to be able type in the numbers.

```

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41264 (Git) Conda
Build date: 2025/04/14 16:28:55
Build type: Release
Branch: main
Hash: 31b0dbaa79da9e0a8d18c009006a2b1c6cd16404
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.0

```

kaiwas, 2025-04-23:
> This does not work. You have to release the Shift key to be able type in the numbers.

Thank you!
This is sad.
If I release the key and the mouse accidentally moves, I lose the direction along the axis. And the XYZ keys do not work

semhustej, 2025-04-23:
> 
> Thank you! This is sad. If I release the key and the mouse accidentally moves, I lose the direction along the axis. And the XYZ keys do not work

Yes, one has to be careful about mouse movement before typing in the value.

tetektoza, 2025-04-23:
@furgo16 not sure if you've meant to tag the person you've tagged, but the behavior I'm fixing in Sketcher is quite different than Draft/BIM for example, and I guess it would require changing in maaany places in the app. They are quite different, but I can look into the above one, although I don't have much clue what you've been discussing so far

semhustej, 2025-04-23:
> > Thank you! This is sad. If I release the key and the mouse accidentally moves, I lose the direction along the axis. And the XYZ keys do not work
> 
> Yes, one has to be careful about mouse movement before typing in the value.

Just to be clear: I would not propose to change this behavior. I don't find this limiting when drafting with FreeCAD.

I checked in BricsCAD (so I guess AutoCAD works the same) and the behavior is similar.

furgo16, 2025-04-23:
@tetektoza oops, sorry, I pinged the wrong person. It was rather a question than a feature request. Thanks for the answer in any case!

kaiwas, 2025-04-23:
> I would not propose to change this behavior. I don't find this limiting when drafting with FreeCAD.

There is no difference how it is done in bricsCAD & others .
We can always do it better than anywhere else.

The current behavior leads to data loss and inaccurate results. If you like to hold your breath while performing actions or release the mouse so as not to accidentally move it - that is your right, but it does not mean that this is the right option.

In addition, some of the tools already work correctly. And it would be good if uniformity was widespread everywhere.

kaiwas, 2025-04-23:
For example, ArchiCAD. I can set the direction with the mouse, then press X or TAB and enter the value. Then I can scratch my head without releasing the mouse and look at the reference book located next to me on the table. And not worry at all about the fact that the entered value will be lost. I am sure that AutoCAD does not behave the same way as FreeCAD. I have not encountered such illogical behavior anywhere else.

[image]

kaiwas, 2025-04-23:
> So can this issue be closed or what is the verdict?

What is the verdict? There is a private opinion that it is possible (it seemed to him) that everything works correctly.
There are two problems at once (locales and interception of focus by shift), and you are thinking whether to close the issue.

kaiwas, 2025-04-23:
There is no such problem in BricksCAD and AutoCAD because there is no need to set the direction along the axis or press Shift. Because there snapping works along the axes and with a step in degrees that can be adjusted. But in FreeCAD it is simply a necessity.

kaiwas, 2025-04-23:
> The X, Y and Z keys are not hard-coded. They can be changed in the Preferences:

Probably it is necessary to explain that working in two locales implies using both. If I assign the Cyrillic layout, the English one will not work for me (I just checked). At the same time, I cannot assign two options at once.

Roy-043, 2025-04-23:
Please also read:
https://wiki.freecad.org/Draft_Snap_Ortho

Roy-043, 2025-04-23:
> You probably haven't noticed, but I have this type of snapping enabled. But it doesn't work if there are no objects.

That is not the case. Try switching off all other snaps.

kaiwas, 2025-04-23:
> That is not the case. Try switching off all other snaps.

First, there is nothing wrong with all bindings being enabled. This should not affect the process (of course, if everything works fine).
Second, it does not work well if I activate bindings with KTRL. I see that the snap works, but when entering values, I cannot enter numbers other than zero. Try it yourself.

https://github.com/user-attachments/assets/38259153-8594-4082-9df3-94ac0beb34f6

Roy-043, 2025-04-23:
> Probably it is necessary to explain that working in two locales implies using both. If I assign the Cyrillic layout, the English one will not work for me (I just checked). At the same time, I cannot assign two options at once.

Users switching locales seems rare. Have you tried using keys that are language independent (`*`, `+`, `=`...)?

kaiwas, 2025-04-23:
> Users switching locales seems rare. Have you tried using keys that are language independent (`*`, `+`, `=`...)?

Yes, it is possible, but how does it work in other programs?

The numbers on the numpad and in the regular keys give different values ​​when shifting (I personally am used to using the numpad), but someone can use a row of numeric keys.
In any case, all this looks like an attempt to find another workaround. This is not good. The user should not look for solutions to problems when working with keys regardless of the layout.

Let's say I make some combination work. But how can I assign two keys at once? I can't specify X,Ч in the settings.

[image]

semhustej, 2025-04-23:
> > Users switching locales seems rare.
> 
> Yes, it is possible, but how does it work in other programs?
> 

Is it common for other software to switch seamlessly between language locales? For instance in Blender if you move object and press Ч, does it lock the movement to the X axis and after you switch to ENG the X key works still the same?

semhustej, 2025-04-23:
> [@semhustej](https://github.com/semhustej) For example, AutoCAD. If I have snapping on axes or with an angle enabled, it is not a problem to stick to the direction. But as soon as I disable angular snapping (freecad option), building a line vertically becomes a problem. The direction is set with SHIFT, but I cannot enter numbers (freecad option). And this is terrible
> 

I agree that having AutoCAD's Polar tracking and Object Snap Tracking would be nice - I find these tools very useful. I think it is very hard to develop these snapping tools and there are probably other areas where development effort would bring more important results.

I also agree that movement of the mouse canceling values that user did put in a field is annoying sometimes, I just don't think the case described here is it.

My test with BricsCAD shows the Shift key for locking directions works exactly the same as FreeCAD.

kaiwas, 2025-04-23:
> after you switch to ENG the X key works still the same?

no. in blender/ corel draw language layout does not affect the action (I use them most of all).
the only difference is in visible symbols in text fields

kaiwas, 2025-10-07:
> Hi!

still relevant