FreeCAD/FreeCAD #20939: FEM: The refactored CalculiX solver object doesn't work on Windows

FEA-eng, 2025-04-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The following error messages appear after enabling the "Pipeline only" checkbox in CalculiX preferences, adding a new ccx solver object and double-clicking on it on Windows (it was tested on Ubuntu before):

```
13:28:49  pyException: Traceback (most recent call last):
  File "E:\Program Files\FreeCAD_weekly-builds-41365-conda-Windows-x86_64-py311\Mod\Fem\femviewprovider\view_solver_calculix.py", line 47, in setEdit
    task = task_solver_calculix._TaskPanel(vobj.Object)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Program Files\FreeCAD_weekly-builds-41365-conda-Windows-x86_64-py311\Mod\Fem\femtaskpanels\task_solver_calculix.py", line 57, in __init__
    self.text_log = self.form.te_output
                    ^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'PySide6.QtWidgets.QWidget' object has no attribute 'te_output'
```

Btw. I think that when this preference is enabled, the refactored solver should be added with the Analysis container when the default solver is CalculiX.

@marioalexis84 FYI

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41365 (Git)
Build date: 2025/04/22 11:42:50
Build type: Release
Branch: main
Hash: 84b57d67b50ac64be2246418fb4b49406904f3a9
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/windowsvista
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-04-25:
@FEA-eng out of curiosity, if you make the changes as per https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:FEM_Ui_file_load on your local Windows build, does it load correctly?

FEA-eng, 2025-04-27:
> [@FEA-eng](https://github.com/FEA-eng) out of curiosity, if you make the changes as per [main...Syres916:FreeCAD:FEM_Ui_file_load](https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:FEM_Ui_file_load) on your local Windows build, does it load correctly?

No, there are still errors.

Syres916, 2025-04-28:
OK, well that's good in a way, as this passes the CI tests and I would hope it's part of the tests then could this not be a packaging issue rather than code specific?

closed 2025-05-15 by FEA-eng