FreeCAD/FreeCAD #20945: Assembly: Crashes when canceling, selecting edges or validating constraints

StenellaCoeruleoalba, 2025-04-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Freecad crashes while using the Assembly Workbench, with an assembly active mainly while selecting edges, canceling, using with Cmd-Z, and while clicking "OK" on a constraint

Using macOS Sequoia with Freecad 1.0

### Full version info

```shell
OS: macOS 15.4.1
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CadbaseLibrary 1.0.3
  * btl 0.9.9
  * dodo 1.0.1
  * addFC 1.0.7
  * Plot 2024.11.26
  * 3DfindIT 1.2.0
  * ProDarkThemePreferencePack 1.0.0
  * BillOfMaterials 1.0.2
  * FreeCAD-themes 2024.7.24
  * freecad.gears 1.3.0
  * FeedsAndSpeeds 0.6.0
  * woodworking 0.21.2.33771.0.23.1
  * OpenTheme 2024.9.1
  * Dracula 0.0.5
  * FreeGrid 2.1.1
  * CfdOF 1.29.4
  * sheetmetal.backup1732441979.0018492 0.5.7 (Disabled)
  * fasteners 0.5.34
  * Rocket 4.0.1
  * Estimate 0.1.2
  * CurvedShapes 1.0.13
  * Ship 2024.11.26
  * ExtremeProDark 2.9.0
  * sheetmetal 0.6.13
  * MakerWorkbench 1.0.1
  * FrameForge 0.1.0
  * ExplodedAssembly
  * Beltrami 1.1.0
  * Curves 0.6.53
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

StenellaCoeruleoalba, 2025-04-23:
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [50470]
Path:                  /Applications/FreeCAD.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-04-23 16:01:12.2221 +0200
OS Version:            macOS 15.4.1 (24E263)
Report Version:        12
Bridge OS Version:     9.4 (22P4248)
Anonymous UUID:        9D07B512-924A-364F-E799-EFBB1363501C

Sleep/Wake UUID:       B98289F4-79B8-4C8D-9871-1B8E655DF719

Time Awake Since Boot: 320000 seconds
Time Since Wake:       6852 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x00000001d0000093
Exception Codes:       0x0000000000000001, 0x00000001d0000093

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [50470]

VM Region Info: 0x1d0000093 is not in any region.  Bytes after previous region: 1225470100  Bytes before following region: 105545331638125
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      VM_ALLOCATE                 186ef0000-186f4d000    [  372K] rw-/rwx SM=PRV  
--->  GAP OF 0x5ffe790b3000 BYTES
      MALLOC_NANO              600000000000-600020000000 [512.0M] rw-/rwx SM=PRV  

Kernel Triage:
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libpython3.11.dylib           	       0x101b448db _PyEval_EvalFrameDefault + 155515
1   libpython3.11.dylib           	       0x1019faff8 _PyFunction_Vectorcall + 504
2   libpython3.11.dylib           	       0x101a021f7 method_vectorcall + 455
3   libFreeCADGui.dylib           	       0x10394eb5a Gui::PythonCommand::isActive() + 298
4   libFreeCADGui.dylib           	       0x103947486 Gui::Command::testActive() + 518
5   libFreeCADGui.dylib           	       0x103953d91 Gui::CommandManager::testActive() + 49
6   libFreeCADGui.dylib           	       0x103d640ab Gui::MainWindow::_updateActions() + 603
7   libQt5Core.5.15.15.dylib      	       0x105deb2ae 0x105c9e000 + 1364654
8   libQt5Core.5.15.15.dylib      	       0x105df0edb QTimer::timeout(QTimer::QPrivateSignal) + 51
9   ???                           	    0x7fe4fa310710 ???

Thread 1:: com.apple.CFSocket.private
0   libsystem_kernel.dylib        	    0x7ff818cda9f6 __select + 10
1   CoreFoundation                	    0x7ff818e2135c __CFSocketManager + 671
2   libsystem_pthread.dylib       	    0x7ff818d13df1 _pthread_start + 99
3   libsystem_pthread.dylib       	    0x7ff818d0f857 thread_start + 15

Thread 2:: com.apple.NSEventThread
0   libsystem_kernel.dylib        	    0x7ff818cd1b4a mach_msg2_trap + 10
1   libsystem_kernel.dylib        	    0x7ff818ce0677 mach_msg2_internal + 83
2   libsystem_kernel.dylib        	    0x7ff818cd8bbe mach_msg_overwrite + 574
3   libsystem_kernel.dylib        	    0x7ff818cd1e3e mach_msg + 19
[... 3495 lines cut ...]
    "path" : "\/Applications\/FreeCAD.app\/Contents\/Resources\/lib\/libvtkParallelDIY-9.3.9.3.dylib",
    "name" : "libvtkParallelDIY-9.3.9.3.dylib"
  },
  {
    "source" : "P",
    "arch" : "x86_64",
    "base" : 5588013056,
    "size" : 61440,
    "uuid" : "9941abf0-0732-30fa-bbb1-162118014d98",
    "path" : "\/Applications\/FreeCAD.app\/Contents\/Resources\/lib\/libdouble-conversion.3.3.0.dylib",
    "name" : "libdouble-conversion.3.3.0.dylib"
  },
  {
    "source" : "P",
    "arch" : "x86_64",
    "base" : 6006177792,
    "size" : 200704,
    "uuid" : "4787948b-9ec2-34fc-a724-fab129bd152e",
    "path" : "\/Applications\/FreeCAD.app\/Contents\/Resources\/lib\/liblz4.1.9.4.dylib",
    "name" : "liblz4.1.9.4.dylib"
  },
  {
    "source" : "P",
    "arch" : "x86_64",
    "bas

luzpaz, 2025-04-23:
@StenellaCoeruleoalba we need an example file. We needs steps to reproduce the crash. We also need someone to test if the crash is reproducible on the development[^1] version. Please stop pasting the whole logfile if nothing has changed from the previous file you have shared.

[^1]: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

StenellaCoeruleoalba, 2025-04-23:
The second log was the result of placing a rivet using a circular edge, clicking OK then using Cmd-Z; then attempting to save.
I've noticed that CMD-Z didn't put the rivet back in it's previous place...
I've sent the file over the discord server

xtemp09, 2025-04-23:
@StenellaCoeruleoalba, the ZIP file you uploaded contains only a single file, while it should contain multiple files.

xtemp09, 2025-04-24:
@StenellaCoeruleoalba, I don't know how to reproduce the crash. Could you record a video?

StenellaCoeruleoalba, 2025-04-24:
Just a minute

StenellaCoeruleoalba, 2025-04-24:
https://upload.disroot.org/r/RooD3xMt#B5r1FPJLDJqNGOUhFaFkAWDulYNoS0Ku+UCsP40KHss=

StenellaCoeruleoalba, 2025-04-24:
Tried on the current 1.1 dev, the bug doesnt occur

maxwxyz, 2025-04-24:
I'm curious if OP cannot reproduce the issue on 1.1 but @xtemp09 you seem to be able to reproduce it on 1.1?

StenellaCoeruleoalba, 2025-04-24:
I will try again

StenellaCoeruleoalba, 2025-04-24:
That's weird, I keep trying but nope, no crashing

xtemp09, 2025-04-25:
I cannot reproduce the crash with the newest weekly bundle and with freshly git-pulled _FreeCAD_ homecompiled in Release mode. However, the assert still fails if _FreeCAD_ compiled in Debug mode.

Below are About infos.

<details>

```ruby
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41365 (Git)
Build date: 2025/04/22 11:42:50
Build type: Release
Branch: main
Hash: 84b57d67b50ac64be2246418fb4b49406904f3a9
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/102.706/1
```

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41354 (Git) AppImage
Build date: 2025/04/21 18:01:03
Build type: Release
Branch: main
Hash: fd390acd53dae91898145a22d74050d0d648a45b
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/95.958/1
```

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41406 (Git)
Build date: 2025/04/24 19:03:18
Build type: Unknown
Branch: main
Hash: d377d43de5e93b2f1124675a313d985b068f5ef5
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.13
shiboken 5.15.13, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/95.958/1
```

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41406 (Git)
Build date: 2025/04/24 19:03:18
Build type: Release
Branch: main
Hash: d377d43de5e93b2f1124675a313d985b068f5ef5
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.13
shiboken 5.15.13, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96.0584/96.0584/1
```

</details>

luzpaz, 2025-05-17:
@StenellaCoeruleoalba Please list the full About info of the 1.1dev version your testing on. TIA

oursland, 2025-08-27:
I can confirm that this issue is affecting me as well.

luzpaz, 2025-08-30:
@oursland is this issue still relevant ?

godiard, 2025-08-30:
I am not sure if is the same issue. or not, but even if not crashing there are logs in the console after the following serquence:
1) Open the assembly demo project
2) Go to the assembly Workbench
3) Open a joint panel by doble click one of the joints in the object tree
4) Close the assembly demo project (without closing the join panel)
5) Reopen the demo assembly project and try to manipulate the assembly or open a joint panel

The console show:
```
<class 'ReferenceError'>: Cannot access attribute 'Name' of deleted object
Traceback (most recent call last):
  File "/home/gonzalo/proyectos/freecad_source/build/debug/Mod/Assembly/JointObject.py", line 1228, in allow
    if self.assembly.Name not in objs_names:

<class 'ReferenceError'>: Cannot access attribute 'Name' of deleted object
Traceback (most recent call last):
  File "/home/gonzalo/proyectos/freecad_source/build/debug/Mod/Assembly/JointObject.py", line 1228, in allow
    if self.assembly.Name not in objs_names:
       ^^^^^^^^^^^^^^^^^^

<class 'ReferenceError'>: Cannot access attribute 'Name' of deleted object
Traceback (most recent call last):
  File "/home/gonzalo/proyectos/freecad_source/build/debug/Mod/Assembly/JointObject.py", line 1882, in addSelection
    moving_part = self.getMovingPart(ref)
                  ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gonzalo/proyectos/freecad_source/build/debug/Mod/Assembly/JointObject.py", line 1867, in getMovingPart
    return UtilsAssembly.getMovingPart(self.assembly, ref)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gonzalo/proyectos/freecad_source/build/debug/Mod/Assembly/UtilsAssembly.py", line 1194, in getMovingPart
    index = names.index(assembly.Name)
```

closed 2025-11-18 by maxwxyz