FreeCAD/FreeCAD #20950: VarSet: Add Property dialog has layout shift when the chosen property type does not accept a value

furgo16, 2025-04-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

On the dialog to add a property to a VarSet, there is an unexpected layout shift when cycling through property types and switching from a property type that expects a value and one that doesn't:

https://github.com/user-attachments/assets/6ca3ca00-83f1-4a61-8017-31f2d21d86fc

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.41399 (Git) Snap 1413
Build date: 2025/04/23 22:19:32
Build type: Release
Branch: main
Hash: 0a3677b9ebe3f6ee9fd97e87ffef454c896566f4
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, boost 1_74, Eigen3 3.4.0, PySide 5.15.11
shiboken 5.15.11, SMESH 7.7.1.0, xerces-c 3.2.3, IfcOpenShell 0.8.0, OCC 7.7.1
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
Installed mods: 
  * Assembly4 0.50.17
  * Launcher
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-04-25:
@pieterhijma FYI

furgo16, 2025-04-25:
Thanks for triaging. I wouldn't say it's a feature though. I feel it's definitely a bug.

pieterhijma, 2025-04-25:
Oh yes, that's annoying and not the intention, so for me this would be a bug as well. I'll try and fix it.

closed 2025-07-25 by kadet1090