FreeCAD/FreeCAD #20956: BIM: error on switching to IFC native mode

furgo16, 2025-04-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:
 
1. Open the BIM example on the Start page
2. Switch to the BIM workbench
3. Activate IFC strict mode by clicking on the IFC toggle button on the status bar
4. Error:

```python
Traceback (most recent call last):
  File "/snap/freecad/1413/usr/Mod/BIM/nativeifc/ifc_status.py", line 234, in on_toggle_lock
    lock_document()
  File "/snap/freecad/1413/usr/Mod/BIM/nativeifc/ifc_status.py", line 423, in lock_document
    ifc_export.direct_conversion(objs, doc)
  File "/snap/freecad/1413/usr/Mod/BIM/nativeifc/ifc_export.py", line 398, in direct_conversion
    prj_obj = ifc_tools.convert_document(doc, silent=True)
  File "/snap/freecad/1413/usr/Mod/BIM/nativeifc/ifc_tools.py", line 141, in convert_document
    document.addProperty("App::PropertyPythonObject", "Proxy", locked=True)
TypeError: 'locked' is an invalid keyword argument for this function
```

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.41399 (Git) Snap 1413
Build date: 2025/04/23 22:19:32
Build type: Release
Branch: main
Hash: 0a3677b9ebe3f6ee9fd97e87ffef454c896566f4
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, boost 1_74, Eigen3 3.4.0, PySide 5.15.11
shiboken 5.15.11, SMESH 7.7.1.0, xerces-c 3.2.3, IfcOpenShell 0.8.0, OCC 7.7.1
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/95.1083/1
Installed mods: 
  * OpenTheme 2024.9.1
  * sheetmetal 0.7.11
  * fasteners 0.5.37
  * fcVM
  * ezydxf
  * Arch
  * Assembly4 0.50.17
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-04-24:
@FlachyJoe this seems to be caused by this change: https://github.com/FreeCAD/FreeCAD/pull/20778/files#diff-2f58fe8ffd31a6d9302296668f6280d27d696e1507b259cc42fde224ef10da0e

Although at a glance I don't understand why it's failing, since the `AddProperty` method does now have the `locked` keyword argument.

furgo16, 2025-04-24:
Oh, I see. It's because the `locked` keyword arg has been added to `DocumentObject::addProperty`, but not to `Document::addProperty`. Does it need to be added there too?

kadet1090, 2025-04-24:
Probably yes, I'll try to make PR with fix for that before next monday - but @FlachyJoe if you will be able to do that before me, feel free to submit PR.

PR #20959 (2025-04-24): Add `locked` keyword to Document::addProperty
Fix #20956

Ping @kadet1090
closed 2025-04-24 by kadet1090