FreeCAD/FreeCAD #20960: [Techdraw] If a page contains  Draft workbench object DXF export command  should show a warnig message

ErtugrulEge, 2025-04-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Greetings to community!

I just learned that if you have a draft workbench object. I created a simple rectangle and added to TD page via
Techdraw->Views from other workbenches->Insert Draft Workbench Object .

Reproducing issue:
1. Open file
2. Open TD page
3. Right click and select Export DXF 
you will get an empty file. 

My suggestion if TD page contains draft workbench object FC should give warning message about you won't have those objects in your DXF file.

[draft_workbench_object_dxf_export.zip](https://github.com/user-attachments/files/19892267/draft_workbench_object_dxf_export.zip)

 in your TD page you can't export it to DXF. 

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41264 (Git) Conda
Build date: 2025/04/14 16:28:55
Build type: Release
Branch: main
Hash: 31b0dbaa79da9e0a8d18c009006a2b1c6cd16404
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/windowsvista
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

markkomartin, 2025-04-24:
Hi @ErtugrulEge .
I tried repeating your instructions with an empty file and everything works.
Then I opened your file and deleted the techdraw view. 
I reinserted techdraw view and the dxf export works fine.

You want to draw a rectangle on draft workbench and then export it to dxf format from techdraw right?

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41365 (Git)
Build date: 2025/04/22 11:42:50
Build type: Release
Branch: main
Hash: 84b57d67b50ac64be2246418fb4b49406904f3a9
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/93.6146/1

[Prove freecad.zip](https://github.com/user-attachments/files/19893572/Prove.freecad.zip)

ErtugrulEge, 2025-04-24:
Hi I know adding rectangle as "view" solves the issue but if you add as "Draft workbench object" FC gives you an empty file. My concern is about user experience because firstly I lost some time to figure out the situation.

luzpaz, 2025-04-24:
Looks related to
* #20626

markkomartin, 2025-04-24:
Ok sorry I misunderstood.
I agree with your suggestion.
The warning message may be : 
"TechDraw can't export an Insert Draft Workbench Object or an Insert Arch Workbench Object to DXF. To export a view as DXF, it must have been created with Insert View "

maxwxyz, 2026-11-07:
This is implemented in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/5c925abed4f32f16028d1a2bab3f602ae9a28d62

Both DXF export commands warn about Draft views being left out and let the user cancel. The Python export API does not prompt.