FreeCAD/FreeCAD #20963: FreeCAD crashes when loading a design with an Assembly made with Assembly4 without having it installed.

leoheck, 2025-04-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

FreeCAD crashes when loading a design with an Assembly made with Assembly4 without having it installed.

The tool should ignore unknown objects/data, loading it partially, or give up loading the file altogether.

But instead, it is crashing miserably.

### Full version info

```shell
Tested on 0.21, 1.0RC4 and 1.1.0devR41354
Using a design made with 1.1-dev.
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-04-24:
The attached SWE was created using ASM4 and does not crash the latest 1.1dev build, the Report View output is:

```
15:32:58  Init: Processing command line files
15:33:12  pyException: <string>(1)<class 'ModuleNotFoundError'>: No module named 'Asm4_libs'
15:33:15  Sketcher::setUpSketch()-T:0.00216765
15:33:15  <PropertyLinks> PropertyLinks.cpp(492): AnimateAssembly_Asm4Ladis#LCS_2.AttachmentSupport missing element reference AnimateAssembly_Asm4Ladis#Sketch_Master ;g47904v2;SKT.Vertex4
15:33:15  <PropertyLinks> PropertyLinks.cpp(492): AnimateAssembly_Asm4Ladis#LCS_2.AttachmentSupport missing element reference AnimateAssembly_Asm4Ladis#Sketch_Master ;g47904;SKT.Edge3
15:33:15  <PropertyLinks> PropertyLinks.cpp(492): AnimateAssembly_Asm4Ladis#LCS_1.AttachmentSupport missing element reference AnimateAssembly_Asm4Ladis#Sketch_Master ;g47902v2;SKT.Vertex2
15:33:15  <PropertyLinks> PropertyLinks.cpp(492): AnimateAssembly_Asm4Ladis#LCS_1.AttachmentSupport missing element reference AnimateAssembly_Asm4Ladis#Sketch_Master ;g47903;SKT.Edge2
15:33:15  <PropertyLinks> PropertyLinks.cpp(492): AnimateAssembly_Asm4Ladis#LCS_1.AttachmentSupport missing element reference AnimateAssembly_Asm4Ladis#Sketch_Master ;g47902;SKT.Edge1
15:33:15  Sketcher::setUpSketch()-T:0.000763027
15:33:15  Sketcher::setUpSketch()-T:0.000658552
15:33:15  Sketcher::Solve()-DogLeg-T:5.3134e-05
```

```
OS: EndeavourOS (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.41402 (Git)
Build date: 2025/04/24 08:51:59
Build type: Release
Branch: main
Hash: a1b22388a8b74fe5757bc767a060102023a5119a
Python 3.13.2, Qt 6.8.3, Coin 4.0.3, Vtk 9.4.1, boost 1_87, Eigen3 3.4.0, PySide 
shiboken , SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Dark/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/96.075/1
Installed mods: 
  * FreeCAD-themes 2025.1.7
  * sheetmetal 0.7.22
  * Reinforcement
  * fasteners 0.5.38
```

[AnimateAssembly_Asm4Ladis.FCStd.zip](https://github.com/user-attachments/files/19893889/AnimateAssembly_Asm4Ladis.FCStd.zip)

leoheck, 2025-04-24:
Ok, I am using a big and complex design, and it may have an issue that breaks on all the versions without the Asm4 installed. I will try to create a smaller project to report this better. 

Oh, I can use your project to start with, @Syres916. Thanks.

leoheck, 2025-04-24:
Ah, now I realize this one is a single file. It could be related to that, then.

@Syres916 indeed, your single file project does not crash FreeCAD without the Asm4 WB in 0.21, 1.0, and 1.1-dev

leoheck, 2025-04-24:
Hm, not a problem with a single external file on a simple Assembly with a single object.

leoheck, 2025-04-24:
Running my big design with the big versions of Freecad, I am seeing this before the crash happens.

<details>
<summary>Crash with FreeCAD 0.21</summary>

```c++
FreeCAD 0.21.2, Libs: 0.21.2R33771 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

NOTICE: OSH Automated Documentation content item  does not support this version of FreeCAD, so is being skipped

Cannot create object 'Origin001': ('App::Point' is not a document object type)
Cannot create object 'Origin003': ('App::Point' is not a document object type)
Cannot create object 'Origin005': ('App::Point' is not a document object type)
Cannot create object 'Origin007': ('App::Point' is not a document object type)
Cannot create object 'Origin009': ('App::Point' is not a document object type)
Cannot create object 'Origin011': ('App::Point' is not a document object type)
Cannot create object 'Origin013': ('App::Point' is not a document object type)
Cannot create object 'Origin015': ('App::Point' is not a document object type)
Cannot create object 'Origin017': ('App::Point' is not a document object type)
Cannot create object 'Origin019': ('App::Point' is not a document object type)
Cannot create object 'Origin021': ('App::Point' is not a document object type)
Cannot create object 'Origin023': ('App::Point' is not a document object type)
Cannot create object 'Origin025': ('App::Point' is not a document object type)
Cannot create object 'Origin027': ('App::Point' is not a document object type)
Cannot create object 'Origin029': ('App::Point' is not a document object type)
Cannot create object 'Origin031': ('App::Point' is not a document object type)
Cannot create object 'Origin033': ('App::Point' is not a document object type)
Cannot create object 'Origin035': ('App::Point' is not a document object type)
Cannot create object 'Origin037': ('App::Point' is not a document object type)
Cannot create object 'Origin039': ('App::Point' is not a document object type)
Cannot create object 'Origin041': ('App::Point' is not a document object type)
Cannot create object 'Origin043': ('App::Point' is not a document object type)
Cannot create object 'Origin045': ('App::Point' is not a document object type)
Cannot create object 'Origin085': ('App::Point' is not a document object type)
<PropertyLinks> PropertyLinks.cpp(759): Lost link to GROUND_SENSOR_Asm Origin001 while loading, maybe an object was not loaded correctly
<PropertyLinks> PropertyLinks.cpp(759): Lost link to GROUND_SENSOR_Asm Origin003 while loading, maybe an object was not loaded correctly
<PropertyLinks> PropertyLinks.cpp(759): Lost link to GROUND_SENSOR_Asm Origin005 while loading, maybe an object was not loaded correctly
<PropertyLinks> PropertyLinks.cpp(759): Lost link to GROUND_SENSOR_Asm Origin007 while loading, maybe an object was not loaded correctly
<PropertyLinks> PropertyLinks.cpp(759): Lost link to GROUND_SENSOR_Asm Origin009 while loading, maybe an object was not loaded correctly
<PropertyLinks> PropertyLinks.cpp(759): Lost link to GROUND_SENSOR_Asm Origin011 while loading, maybe an object was not loaded correctly
<PropertyLinks> PropertyLinks.cpp(759): Lost link to GROUND_SENSOR_Asm Origin013 while loading, maybe an object was not loaded correctly
<PropertyLinks> PropertyLinks.cpp(759): Lost link to GROUND_SENSOR_Asm Origin015 while loading, maybe an object was not loaded correctly
<PropertyLinks> PropertyLinks.cpp(759): Lost link to GROUND_SENSOR_Asm Origin017 while loading, maybe an object was not loaded correctly
<PropertyLinks> PropertyLinks.cpp(759): Lost link to GROUND_SENSOR_Asm Origin019 while loading, maybe an object was not loaded correctly
<PropertyLinks> PropertyLinks.cpp(759): Lost link to GROUND_SENSOR_Asm Origin021 while loading, maybe an object was not loaded correctly
<PropertyLinks> PropertyLinks.cpp(759): Lost link to GROUND_SENSOR_Asm Origin023 while loading, maybe an object was not loaded correctly
<PropertyLinks> PropertyLinks.cpp(759): Lost link to GROUND_SENSOR_Asm Origin025 while loading, maybe an object was not loaded correctly
<PropertyLinks> PropertyLinks.cpp(759): Lost link to GROUND_SENSOR_Asm Origin027 while loading, maybe an object was not loaded correctly
<PropertyLinks> PropertyLinks.cpp(759): Lost link to GROUND_SENSOR_Asm Origin029 while loading, maybe an object was not loaded correctly
<PropertyLinks> PropertyLinks.cpp(759): Lost link to GROUND_SENSOR_Asm Origin031 while loading, maybe an object was not loaded correctly
<PropertyLinks> PropertyLinks.cpp(759): Lost link to GROUND_SENSOR_Asm Origin033 while loading, maybe an object was not loaded correctly
<PropertyLinks> PropertyLinks.cpp(759): Lost link to GROUND_SENSOR_Asm Origin035 while loading, maybe an object was not loaded correctly
<PropertyLinks> PropertyLinks.cpp(759): Lost link to GROUND_SENSOR_Asm Origin037 while loading, maybe an object was not loaded correctly
<PropertyLinks> PropertyLinks.cpp(759): Lost link to GROUND_SENSOR_Asm Origin039 while loading, maybe an object was not loaded correctly
[... cut ...]
No extension found of type 'Gui::ViewProviderFaceTexture'
No extension found of type 'Gui::ViewProviderFaceTexture'
No extension found of type 'Gui::ViewProviderFaceTexture'
No extension found of type 'Gui::ViewProviderFaceTexture'
No extension found of type 'Gui::ViewProviderFaceTexture'
No extension found of type 'Gui::ViewProviderFaceTexture'
No extension found of type 'Gui::ViewProviderFaceTexture'
No extension found of type 'Gui::ViewProviderFaceTexture'
No extension found of type 'Gui::Vie

leoheck, 2025-04-24:
<details>
<summary>Crash with FreeCAD 1.1.0, Libs: 1.1.0devR41354 (Git)</summary>

```c++
Case/v1.3/Design …
➜ freecad_weekly_apprun SENSOR_Asm.FCStd          
FreeCAD 1.1.0, Libs: 1.1.0devR41354 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

OSHAutoDoc workbench (v0.2.3) loaded
DiffTools workbench 0.1.0 loaded
<class 'ModuleNotFoundError'>: No module named 'Asm4_libs'
[REMOVED_MANY_OF_THE_SAME_LOG_ABOVE]
<class 'ModuleNotFoundError'>: No module named 'Asm4_objects'
[REMOVED_MANY_OF_THE_SAME_LOG_ABOVE]
<ElementMap> ElementMap.cpp(483): No hasherRef
<ElementMap> ElementMap.cpp(483): No hasherRef
<class 'ModuleNotFoundError'>: No module named 'Asm4_libs'
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45330) [0x7320ca245330]
#1  0x7320cc5d4090 in App::PropertyLink::getValue() const from /home/lheck/Applications/FreeCAD-41354/squashfs-root/usr/bin/../lib/libFreeCADApp.so+0
#2  0x7320cc57bd14 in App::LinkBaseExtension::syncElementList() from /home/lheck/Applications/FreeCAD-41354/squashfs-root/usr/bin/../lib/libFreeCADApp.so+0x3e4
#3  0x7320cc580c1a in App::LinkBaseExtension::update(App::DocumentObject*, App::Property const*) from /home/lheck/Applications/FreeCAD-41354/squashfs-root/usr/bin/../lib/libFreeCADApp.so+0x98a
#4  0x7320cc583784 in App::LinkBaseExtension::onExtendedDocumentRestored() from /home/lheck/Applications/FreeCAD-41354/squashfs-root/usr/bin/../lib/libFreeCADApp.so+0x1c4
#5  0x7320cc47197c in App::DocumentObject::onDocumentRestored() from /home/lheck/Applications/FreeCAD-41354/squashfs-root/usr/bin/../lib/libFreeCADApp.so+0x4c
#6  0x7320cc45a93a in App::Document::afterRestore(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool) from /home/lheck/Applications/FreeCAD-41354/squashfs-root/usr/bin/../lib/libFreeCADApp.so+0x65a
#7  0x7320cc45bc34 in App::Document::afterRestore(bool) from /home/lheck/Applications/FreeCAD-41354/squashfs-root/usr/bin/../lib/libFreeCADApp.so+0x54
#8  0x7320cc67baa8 in App::Application::openDocuments(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >*, App::DocumentInitFlags) from /home/lheck/Applications/FreeCAD-41354/squashfs-root/usr/bin/../lib/libFreeCADApp.so+0x17b8
#9  0x7320cc67cd04 in App::Application::openDocument(char const*, App::DocumentInitFlags) from /home/lheck/Applications/FreeCAD-41354/squashfs-root/usr/bin/../lib/libFreeCADApp.so+0xf4
#10  0x7320cc67dade in App::Application::processFiles(std::__cxx11::list<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&) from /home/lheck/Applications/FreeCAD-41354/squashfs-root/usr/bin/../lib/libFreeCADApp.so+0x2ee
#11  0x7320cdaac09e in Gui::MainWindow::delayedStartup() from /home/lheck/Applications/FreeCAD-41354/squashfs-root/usr/bin/../lib/libFreeCADGui.so+0x19e
#12  0x7320ca8d25db in QObject::event(QEvent*) from /home/lheck/Applications/FreeCAD-41354/squashfs-root/usr/bin/../lib/libQt6Core.so.6+0x23b
#13  0x7320cbba40f4 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /home/lheck/Applications/FreeCAD-41354/squashfs-root/usr/bin/../lib/libQt6Widgets.so.6+0x84
#14  0x7320cd5a02b9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/lheck/Applications/FreeCAD-41354/squashfs-root/usr/bin/../lib/libFreeCADGui.so+0xc9
#15  0x7320ca94d29a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /home/lheck/Applications/FreeCAD-41354/squashfs-root/usr/bin/../lib/libQt6Core.so.6+0x16a
#16  0x7320ca97c120 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /home/lheck/Applications/FreeCAD-41354/squashfs-root/usr/bin/../lib/libQt6Core.so.6+0x320
#17  /home/lheck/Applications/FreeCAD-41354/squashfs-root/usr/bin/../lib/libQt6Core.so.6(+0xf3604) [0x7320ca6f3604]
#18  /home/lheck/Applications/FreeCAD-41354/squashfs-root/usr/bin/../lib/./libglib-2.0.so.0(+0x5b5e6) [0x7320c8fbf5e6]
#19  /home/lheck/Applications/FreeCAD-41354/squashfs-root/usr/bin/../lib/./libglib-2.0.so.0(+0x5e7b7) [0x7320c8fc27b7]
#20  /home/lheck/Applications/FreeCAD-41354/squashfs-root/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7320c8fc2ece]
#21  0x7320ca6efd0e in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /home/lheck/Applications/FreeCAD-41354/squashfs-root/usr/bin/../lib/libQt6Core.so.6+0x6e
#22  0x7320ca9857fc in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /home/lheck/Applications/FreeCAD-41354/squashfs-root/usr/bin/../lib/libQt6Core.so.6+0x1ac
[... cut ...]
#24  0x7320cd4fb414 in Gui::Application::runApplication() from /home/lheck/Applications/FreeCAD-41354/squashfs-root/usr/bin/../lib/libFreeCADGui.so+0x734
#25  /home/lheck/Applications/FreeCAD-41354/squashfs-root/usr/bin/freecad(+0x6eb8) [0x626c13be5eb8]
#26  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x7320ca22a1ca]
#27  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x7320ca22a28b]
#28  /home/lheck/Applications/FreeCAD-41354/squashfs-root/usr/bin/freecad(+0x7242) [0x626c13be6242]

```
</details>

leoheck, 2025-04-24:
They all crash with the same message:

```c++
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45330) [0x7decce645330]
#1  0x7decd09883a0 in App::PropertyLink::getValue() const from /tmp/.mount_FreeCAJfg0j9/usr/bin/../lib/libFreeCADApp.so+0
#2  0x7decd0934934 in App::LinkBaseExtension::syncElementList() from /tmp/.mount_FreeCAJfg0j9/usr/bin/../lib/libFreeCADApp.so+0x4b4
#3  0x7decd0939812 in App::LinkBaseExtension::update(App::DocumentObject*, App::Property const*) from /tmp/.mount_FreeCAJfg0j9/usr/bin/../lib/libFreeCADApp.so+0x8d2
#4  0x7decd093b7cc in App::LinkBaseExtension::onExtendedDocumentRestored() from /tmp/.mount_FreeCAJfg0j9/usr/bin/../lib/libFreeCADApp.so+0x1cc
#5  0x7decd0848144 in App::DocumentObject::onDocumentRestored() from /tmp/.mount_FreeCAJfg0j9/usr/bin/../lib/libFreeCADApp.so+0xe4
#6  0x7decd082e479 in App::Document::afterRestore(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool) from /tmp/.mount_FreeCAJfg0j9/usr/bin/../lib/libFreeCADApp.so+0x819
#7  0x7decd082f5bd in App::Document::afterRestore(bool) from /tmp/.mount_FreeCAJfg0j9/usr/bin/../lib/libFreeCADApp.so+0xbd
#8  0x7decd0a13411 in App::Application::openDocuments(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >*, bool) from /tmp/.mount_FreeCAJfg0j9/usr/bin/../lib/libFreeCADApp.so+0x1c41
#9  0x7decd0a13c08 in App::Application::openDocument(char const*, bool) from /tmp/.mount_FreeCAJfg0j9/usr/bin/../lib/libFreeCADApp.so+0xc8
#10  0x7decd0a16789 in App::Application::processFiles(std::__cxx11::list<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&) from /tmp/.mount_FreeCAJfg0j9/usr/bin/../lib/libFreeCADApp.so+0x169
#11  0x7decd196f68d in Gui::MainWindow::delayedStartup() from /tmp/.mount_FreeCAJfg0j9/usr/bin/../lib/libFreeCADGui.so+0xad
#12  0x7deccecde190 in QObject::event(QEvent*) from /tmp/.mount_FreeCAJfg0j9/usr/bin/../lib/libQt5Core.so.5+0x260
#13  0x7deccfb7035e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAJfg0j9/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#14  0x7decd14f7129 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAJfg0j9/usr/bin/../lib/libFreeCADGui.so+0x99
#15  0x7deccecb3fc8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAJfg0j9/usr/bin/../lib/libQt5Core.so.5+0x108
#16  0x7deccecb6c91 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /tmp/.mount_FreeCAJfg0j9/usr/bin/../lib/libQt5Core.so.5+0x181
#17  /tmp/.mount_FreeCAJfg0j9/usr/bin/../lib/libQt5Core.so.5(+0x308cb3) [0x7decced08cb3]
#18  /tmp/.mount_FreeCAJfg0j9/usr/bin/../lib/./libglib-2.0.so.0(+0x5a417) [0x7deccd311417]
#19  /tmp/.mount_FreeCAJfg0j9/usr/bin/../lib/./libglib-2.0.so.0(+0x5d3c7) [0x7deccd3143c7]
#20  /tmp/.mount_FreeCAJfg0j9/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7deccd3149be]
#21  0x7decced083d6 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAJfg0j9/usr/bin/../lib/libQt5Core.so.5+0x66
#22  0x7deccecb2bbb in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAJfg0j9/usr/bin/../lib/libQt5Core.so.5+0x11b
#23  0x7deccecba6a2 in QCoreApplication::exec() from /tmp/.mount_FreeCAJfg0j9/usr/bin/../lib/libQt5Core.so.5+0x82
#24  0x7decd145bddc in Gui::Application::runApplication() from /tmp/.mount_FreeCAJfg0j9/usr/bin/../lib/libFreeCADGui.so+0x1ecc
#25  /tmp/.mount_FreeCAJfg0j9/usr/bin/freecad(+0x7620) [0x5d07ef573620]
#26  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x7decce62a1ca]
#27  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x7decce62a28b]
#28  /tmp/.mount_FreeCAJfg0j9/usr/bin/freecad(+0x7981) [0x5d07ef573981]
```

Zolko-123, 2025-04-24:
```
#1  0x7decd09883a0 in App::PropertyLink::getValue() const from /tmp/.mount_FreeCAJfg0j9/usr/bin/../lib/libFreeCADApp.so+0
#2  0x7decd0934934 in App::LinkBaseExtension::syncElementList() from /tmp/.mount_FreeCAJfg0j9/usr/bin/../lib/libFreeCADApp.so+0x4b4
#3  0x7decd0939812 in App::LinkBaseExtension::update(App::DocumentObject*, App::Property const*) from /tmp/.mount_FreeCAJfg0j9/usr/bin/../lib/libFreeCADApp.so+0x8d2
#4  0x7decd093b7cc in App::LinkBaseExtension::onExtendedDocumentRestored() from /tmp/.mount_FreeCAJfg0j9/usr/bin/../lib/libFreeCADApp.so+0x1cc
#5  0x7decd0848144 in App::DocumentObject::onDocumentRestored() from /tmp/.mount_F...
```

What Asm4-type object did you use ? Patterns (linear or polar), variantLink, Mirror ?

leoheck, 2025-04-25:
@Zolko-123, I could track this problem to a single sub-assembly.
It is particularly difficult to track the issue since there is no indication of which file/object the problem arises in.

This sub-assembly in particular uses Linear Arrays.

And I noticed that something strange happens when using a linear array of another linear array.

It duplicates the objects, see this picture. The last 2 items are replicated (the hidden objects). If I hide one that is selected, for instance, it hides the replica too, and vice versa.

[image]

leoheck, 2025-04-25:
I could not replicate the crash on a small project using a linear array of another linear array.
But they are well-formed.
I will try to recreate the arrays above to see if this gets improved.

leoheck, 2025-04-25:
Moving the problematic arrays to the root of the Assembly, with the parts that are duplicated, fixes this problem automatically. I don't have to recreate the arrays.

leoheck, 2025-04-25:
Alright, fixing this thing makes Freecad not crash when Asm4 WB is not installed.

leoheck, 2025-04-25:
I could fix the crash by fixing the Arrays, but I could not reproduce the crash on a small file.

FEA-eng, 2025-10-07:
@leoheck Is this issue still relevant or can it be closed?

leoheck, 2025-10-07:
Nope, it is not crashing anymore.

```
Version: 1.1.0dev.14555 (Git shallow)
```