FreeCAD/FreeCAD #20964: TechDraw: User created template regression

Syres916, 2025-04-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Using https://wiki.freecad.org/TechDraw_TemplateGenerator#Resulting_macro the svg output is:

```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>

<svg
  xmlns="http://www.w3.org/2000/svg" version="1.1"
  xmlns:freecad="http://www.freecad.org/wiki/index.php?title=Svg_Namespace"
  width ="420mm"
  height="297mm"
  viewBox="0 0 420 297">
    <g id="drawing-frame"
      style="fill:none;stroke:#000000;stroke-width:0.5;stroke-linecap:round">
      <rect width="390" height="277" x="20" y="10" />
      <rect width="400" height="287" x="15" y="5" />
    </g>

    <g id="titleblock"
      transform="translate(230,287)">
      <g id="titleblock-frame"
        style="fill:none;stroke:#000000;stroke-width:0.35;stroke-linecap:miter;stroke-miterlimit:4">
        <path d="m   0,  0 l   0,-63" />
        <path d="m   0,-63 l 180,  0" />
        <path d="m   0,-30 h 155" />
        <path d="m 155,  0 v -63" />
      </g>
      <g id="titleblock-text-non-editable"
        style="font-size:5.0;text-anchor:start;fill:#000000;font-family:osifont">
        <text x="  4.5" y="-43.5 ">Some static text</text>
        <text x="  4.5" y="-13.5 ">More static text</text>
        <text x="162.5" y="-3.5 " transform="rotate(-90,162.5,-3.5 )">Vertical static text</text>
      </g>
    </g>

    <g id="titleblock-editable-texts"
      style="font-size:7.0;text-anchor:start;fill:#0000d0;font-family:osifont">
      <text freecad:editable="EdiText-1" x="290" y="243.5">  <tspan>Some editable text</tspan>  </text>
      <text freecad:editable="EdiText-2" x="290" y="273.5">  <tspan>More editable text</tspan>  </text>
      <text freecad:editable="EdiText-3" x="403" y="282.5" transform="rotate(-90,403,282.5)">  <tspan>90° editable text</tspan>  </text>
    </g>

</svg>
```

Prior to #20220 the output was:

[image]

Post #20220 it is now:

[image]

Originally reported as a possible Python version issue https://forum.freecad.org/viewtopic.php?t=96562 that was discounted as a coincidence with the merge of the PR referenced.

### Full version info

```shell
OS: EndeavourOS (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.41402 (Git)
Build date: 2025/04/24 08:51:59
Build type: Release
Branch: main
Hash: a1b22388a8b74fe5757bc767a060102023a5119a
Python 3.13.2, Qt 6.8.3, Coin 4.0.3, Vtk 9.4.1, boost 1_87, Eigen3 3.4.0, PySide 
shiboken , SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Dark/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/96.075/1
Installed mods: 
  * FreeCAD-themes 2025.1.7
  * sheetmetal 0.7.22
  * Reinforcement
  * fasteners 0.5.38
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

FC-FBXL5, 2025-04-24:
The problem in short:
Only one click handle running from left to right is displayed, the active area above is huge.
And only one (of several) editable text is selectable, it is the last created one in this example (and I can confirm with other examples).

WandererFan, 2025-05-03:
Based on initial investigation, it appears the new QGSvgRenderer-based algo depends on text and tspan elements having an id attribute.  All of the QSvgRenderer methods (boundsOnElement(id), transformForElement(id), etc) return garbage if the id is not correct.  

Potential solutions: 
a) edit all the templates so text and tspan elements have unique ids.  Not sure that this would actually work as AFAICT tspan does not support an id attribute.   
- or - 
b) replace the new QSvgRenderer algo with the xml parsing approach from cancelled PR #19711.

FlachyJoe, 2025-05-04:
@benj5378 what's your opinion?

benj5378, 2025-05-04:
> [@benj5378](https://github.com/benj5378) what's your opinion?

Generating ids for elements missing would be a simple and easy fix

WandererFan, 2025-05-04:
added id attributes to the 3 freecad::editables in the sample template.  1 & 2 are OK, but the rotated one does not rotate.

[image]

text #3 has a transform clause for rotation, but from the docs on QSvgRenderer::transformForElement: "The transformation of the element itself is not included." 

The algo needs to parse the current element, find the transform and combine it with the value from transformForElement.

WandererFan, 2025-05-04:
"Generating ids for elements missing would be a simple and easy fix."

are you volunteering?

benj5378, 2025-05-04:
> The transformation of the element itself is not included

???? That makes it basically useless. Why would they do that?

benj5378, 2025-05-04:
> "Generating ids for elements missing would be a simple and easy fix."
> 
> are you volunteering?

If no one else are, should be very easy

FC-FBXL5, 2025-05-04:
Do you mean the algo that reads the template file automatically adds IDs to the embedded template code?

I would prefer to have IDs in tags optional rather than mandatory. To give an svg file a readable structure it is sufficient to have IDs In group tags. Giving every grouped element a separate tag seems quite useless to me.

But at least I can confirm that templates are useable again if the text tags hold **different** IDs. And the click handle for rotated text has never rotated, but I could live with that.

FC-FBXL5, 2025-05-04:
What is meant by "...we remove all tspan due to Qt bug..." ?

The namespace wiki page says:
"Default text has to be surrounded by tspan tags otherwise the displayed text within the template/symbol will not synchronise with modified variable content."

Have you already found a way to synchronise the displayed text without the need of span tags?

benj5378, 2025-05-04:
> Do you mean the algo that reads the template file automatically adds IDs to the embedded template code?

Yes

> I would prefer to have IDs in tags optional rather than mandatory. To give an svg file a readable structure it is sufficient to have IDs In group tags. Giving every grouped element a separate tag seems quite useless to me.

Qt want every `text` tag to have an `id`, there is no way around that. If adding `id` fixes the bug, we should not abstain it, just because it looks ugly? (It's not even visual)

> "Default text has to be surrounded by tspan tags otherwise the displayed text within the template/symbol will not synchronize with modified variable content."
> 
> Have you already found a way to synchronize the displayed text without the need of span tags?

I'm not sure what it means, but either I have found a way to synchronize without, or I have introduced a regression, since all `tspan` are removed upon processing.

FC-FBXL5, 2025-05-05:
> Qt wants every text tag to have an id, there is no way around that.
This is a new "feature" of Qt6, isn't it? it was no problem before.

> If adding id fixes the bug, we should not abstain it, just because it looks ugly?
Ugly or  not, it is more useless information to type, either for me or my macros.
The more important question is: Do we need individual IDs, or can we use the same dummy ID for all text tags?

Some useful descriptions dealing with template and symbol texts need a major update once the PR is merged and working as promised.

BTW - Where can I find information about the algorithm that deals with freecad:editable and freecad:autofill, or how to add other elements to FreeCAD's namespace?

benj5378, 2025-05-05:
> Do we need individual IDs, or can we use the same dummy ID for all text tags?

Individual ids, but the user don't need to add then, it will be done automatically.

closed 2025-05-29 by WandererFan

mpfusion, 2025-06-04:
I updated to `Version: 1.1.0dev.42014 (Git)` and it's still the same issue, none of the fields are editable:

[image]
Template is `A4_LandscapeTD.svg`.

kris969, 2025-06-04:
I confirm it still not working on the current 1.1.0 dev version.

It works when I switch back to stable release v1.0.

Here is my setup:

OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.41905 (Git) Snap 1474
Build date: 2025/05/30 18:02:58
Build type: Release
Branch: main
Hash: 239a05b242b9162fd7a5ab13dcfa86b097c5ad55
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, boost 1_74, Eigen3 3.4.0, PySide 5.15.11
shiboken 5.15.11, SMESH 7.7.1.0, xerces-c 3.2.3, IfcOpenShell 0.8.0, OCC 7.7.1
Locale: French/France (fr_FR)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 139/140.007/1
Installed mods: 
  * symbols_library
  * offline-documentation 1.0.0-alpha
  * parts_library

WandererFan, 2025-06-04:
The fix in PR #21131 only works on Qt6.

mpfusion, 2025-06-04:
Will there be a fix for Qt5?

kris969, 2025-06-04:
I am running Ubuntu 25.04, I was thinking this version using qt6. Am I wrong?
When I check from FreeCAD, version of QT used is Qt 5.15.15
 
Where this QT library is coming from? I suppose in the snap witch the only way to access to dev branch on Linux. isn't it?

FC-FBXL5, 2025-06-04:
For me the fix works excellent since weekly build 41905.

tspan tags are still needed in templates. Shouldn't they have become obsolete with the fix?

> ...as we remove all tspan due to Qt bug...

kris969, 2025-06-05:
@WandererFan I normally use snap beta version on Linux Ubuntu.
I prefer to use the dev branch, it's for me a way to contribute to this project and to give my feedback.

I tried the Appimage, and if this case it works, but when I have a look on the package used, it also use Qt 5.
I have the same behaviour when I use the released snap version 1.0.
In both cases it works well with Qt5.

So I'm not sure Qt6 must be mandatory, and if it is the case the snap version should be using Qt 6.

here is what I get with appimage.
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, **Qt 5.15.15**, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

PR #21131 (2025-05-29): TechDraw: fix incorrect transform of editable fields

## Issues

Fixes #20964
commit 9709fc43 (2025-05-29): TechDraw: fix incorrect click handles of editable labels (#21131)

Fixes #20964:
- When `text` element doesn't have id
- When `text` element has transform attribute