FreeCAD/FreeCAD #20968: TechDraw_GeometricHatch does not enable scale smaller than 0.1 during hatch creation

semhustej, 2025-04-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

- open file (remove the .zip extension)

[FC-scale-testing .FCStd.zip](https://github.com/user-attachments/files/19895718/FC-scale-testing.FCStd.zip)

- select the face on the page
- start T`echDraw_GeometricHatch`
- set `Scale` to 0.01
- the Scale gets set back to 1. It is not possible to set the value lower than 0.1

See behavior in the video:

https://github.com/user-attachments/assets/8c7ad71a-d819-421d-8c79-91906083b7e3

**Expected behavior:**
The Scale value should be possible to be set to the smallest decimal value available in the dialogue

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41365 (Git)
Build date: 2025/04/22 11:42:50
Build type: Release
Branch: main
Hash: 84b57d67b50ac64be2246418fb4b49406904f3a9
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.0
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-04-29:
I believe the reason for the task panel restriction is the performance impact on a heavy model if the user accidentally keeps clicking the down arrow on the spinbox. Here's an example mechanical handling drawing with Diagonal5 (commonly used in mech drawings) set to 0.1 and it's already laggy (this is a relatively small area of a A0 drawing):

[image]

semhustej, 2025-04-29:
> I believe the reason for the task panel restriction is the performance impact on a heavy model if the user accidentally keeps clicking the down arrow on the spinbox. Here's an example mechanical [...]

I thought this might be the reason, but in my opinion limiting scale should not be solution. In AEC applications, it is not unique to use hatch scales such as 1000, 0.001. One example can be user in this forum thread: https://forum.freecad.org/viewtopic.php?t=96538

The scale depends on how the hatch is defined and also desire output scale. The current behavior is quite confusing and looks like a bug to a new user. Not being able to type in a number which is offered by FreeCAD to be changed occurs as a bad user experience to me.

OFF-TOPIC:
I ran into performance issues using too dense hatch patterns. I think the better solution for that would be a warning dialogue after a timeout (for instance 15-30 seconds). User would be warned that the calculation might take long time. This is how it is solved in AutoCAD and it's clones.