FreeCAD/FreeCAD #20969: BIM: W, P keyboard shortcut is not available to select plane

furgo16, 2025-04-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Switching working planes is a key and very often used operation when working on BIM models. So much that it deserves its own keyboard shortcut for a more efficient workflow. The Draft workbench does have the <kbd>W</kbd>, <kbd>P</kbd> shortcut for that, but it does not seem to be available when switching to the BIM workbench.

According to https://github.com/FreeCAD/FreeCAD/issues/20958#issuecomment-2828008933

> AFAICT the `W`, `P` shortcut does not work in the BIM WB because there is no `Draft_SelectPlane` menu item.

### Full version info

```shell
1.1.0dev
```

### Subproject(s) affected?

BIM

### Anything else?

Is it a matter of simply adding a "Select working plane" menu entry with shortcut to BIM's "Utils" menu, or is there a better way?

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-04-24:
Related issue: https://github.com/FreeCAD/FreeCAD/issues/14825

ErtugrulEge, 2025-04-24:
In general, BIM has some features of Draft,  like lines,rectangles,move,copy tools etc but also does not have some features like Working plane proxy or draft edit tool. You can draw a line in BIM, but you have to go back draft to edit line.  Same problem, D-E (Draft edit shortcut) works just in draft. 

In short, BIM should include all of draft workbench features not some of them or we hosuld explicitly seperate BIM and Draft but this is against the architectural work flow.

kaiwas, 2025-04-24:
> In short, BIM should include all of draft workbench features not some of them or we should explicitly seperate BIM and Draft but this is against the architectural work flow. A regulation is needed

+1
Although I should add that I have to switch to DRAFT for arrays, which for some reason are not in BIM. For example, a link array by points.

furgo16, 2025-05-06:
@kaiwas arrays seem to be on the BIM toolbar. Are you looking for a particular type of array that's missing?

[image]

> does not have some features like Working plane proxy

@ErtugrulEge I agree with you. If it helps, for this particular command, it is already available in the BIM workbench, via the Utils menu, or via the BIM Views panel context menu. There is no shortcut for it, though.

kaiwas, 2025-05-06:
> arrays seem to be on the BIM toolbar. Are you looking for a particular type of array that's missing?

Among the arrays in the DRAFT workbench there are reference arrays. For some reason they are not in BIM, although they have more capabilities. 

[image]

furgo16, 2025-05-06:
Unrelated to this issue, but to address this comment, I think it might be worth submitting a PR that:
- Adds the missing array commands to the BIM toolbar
- Groups all array commands in a drop-down, just like in Draft

kadet1090, 2025-05-12:
Merged via https://github.com/FreeCAD/FreeCAD/pull/21282

closed 2025-05-12 by kadet1090

kaiwas, 2025-05-14:
It seems I wrote in the wrong place. Apologies.

https://github.com/FreeCAD/FreeCAD/pull/21157#issuecomment-2880296425

kaiwas, 2025-05-14:
Maybe WPX/WPY/WPZ ?

furgo16, 2025-05-14:
> @furgo16 I have a key conflict for the WP1/WP2/WP3 combinations

This PR did not modify those key combinations, they were already there before this got merged. It might be worth reporting it as a new issue.

kaiwas, 2025-05-14:
> It might be worth reporting it as a new issue.

ok. thanks for the info