FreeCAD/FreeCAD #20978: [Start WB & thumbnails] FreeCAD-41365 and f3d are sucking up cycles and heating up CPU  (67 C)

pixelwaster, 2025-04-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When starting the latest weeklies (1.1dev-41365) on Win10, I get a log screen from f3d with an OK to continue button. My computer then starts to lag and the area near where I rest my hand start to get warmer. A quick look at the tool bar shows the CPU temp at 70C and looking at the task manager (always running) showed 5 f3d.exe running under FreeCAD each at 19%. (CPU is at 100)

I immediately kill all five of them and the temp drops to its normal 32-38C and the computer becomes responsive again.  

I have not had issues with f3d since updating to 3.1
Why was FreeCAD calling f3d? 
Why was I having overheating issues again? New fan and thermal paste fixed them all months ago. 

Looking though the merged pull requests I found my answer in #19489 Start: create thumbnails using f3d

Is there a way to limit thumbnail generation, especially on larger files? Or only use a single f3d or ver 3?

Sidenote:
I watch  my temps and cycle usage closely because Microsoft has a tendency to restart things that are not well written and use up all the memory or a couple of COM Surrogate freak out and try and start my computer on fire.

[image]

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41365 (Git)
Build date: 2025/04/22 11:42:50
Build type: Release
Branch: main
Hash: 84b57d67b50ac64be2246418fb4b49406904f3a9
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenLight.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/112.198/1
Installed mods: 
  * AddonManager
  * Assembly
  * BIM
  * btl 0.9.9
  * CAM
  * CurvedShapes 1.0.13
  * Curves 0.6.61
  * Draft
  * DynamicData 2.74.0
  * Fem
  * FreeCAD-themes 2025.1.7
  * freecad.gears 1.3.0
  * GrabBag
  * Help
  * IconThemes
  * Idf
  * Import
  * Inspection
  * lattice2 1.0.0
  * Material
  * Measure
  * Mesh
  * MeshPart
  * OpenSCAD
  * OpenTheme 2024.9.1
  * Part
  * PartDesign
  * Plot
  * Points
  * ReverseEngineering
  * Robot
  * sheetmetal 0.7.22
  * Show
  * Sketcher
  * Spreadsheet
  * Start
  * Surface
  * TechDraw
  * Test
  * Tux
  * Web
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-04-25:
@chennes likely from https://github.com/FreeCAD/FreeCAD/pull/19489 could you take a look?

chennes, 2025-04-25:
We are already throttling `f3d` so it's a simple change to decrease the max size it will consider -- I hate to make it yet another user parameter, though. I could also limit the number of simultaneous processes, right now I'm just letting `QThreadPool` deal with that level of detail, but it's not particularly conservative in terms of maxing out your CPUs. Finally, I don't know why it's running an old version of f3d if you have a newer one installed -- was it an intentional choice to keep both on your system, @pixelwaster?

pixelwaster, 2025-04-26:
>  Finally, I don't know why it's running an old version of f3d if you have a newer one installed -- was it an intentional choice to keep both on your system, @pixelwaster? 

 _Originally posted by @chennes in [#20978](https://github.com/FreeCAD/FreeCAD/issues/20978#issuecomment-2830980269)_

I only have ver3.1 It was the ver2,x that was maxing my cpu.

chennes, 2025-04-26:
> I agree with the author of this issue. What if the files are 500 MiB each? I used to open STL files about this size.

We currently limit the size considered to 100MB, so your 500 MiB files will be ignored already. The questions are: should that limit be lower (or user-configurable), and/or should we prevent use of `f3d` v2.x entirely. Are you experiencing problems as well, @xtemp09?

xtemp09, 2025-04-26:
> Are you experiencing problems as well, [@xtemp09](https://github.com/xtemp09)?

No.

----

Files may be stored in a network folder or being synced through services like _Dropbox_. Does the thumbnailer have any safeguards in these situations?

chennes, 2025-04-26:
There's a timeout per-process, so if the resource takes a long time to load that `f3d` process will get killed. Do you have an idea for a different way of addressing it?

xtemp09, 2025-04-27:
> There's a timeout per-process, so if the resource takes a long time to load that `f3d` process will get killed. Do you have an idea for a different way of addressing it?

This idea is ― _FreeCAD_ adds a thumbnail in the beginning of the file during saving. In this case there is no need to load the file to render it to create a thumbnail.

Matroska/MP4/MP3 have cover art that is used as thumbnail. I have never seen that software on my computer stumbled to show a thumbnail of 600 MiB FLAC.

chennes, 2025-04-28:
> This idea is ― _FreeCAD_ adds a thumbnail in the beginning of the file during saving. In this case there is no need to load the file to render it to create a thumbnail.

`f3d` is not used on FreeCAD files (or on image files), only on files that FreeCAD can read, but that it doesn't know how to get a thumbnail for otherwise (e.g. STEP files).

xtemp09, 2025-04-28:
So, if a user opens several STEP files, _FreeCAD_ loads them all and renders their thumbnails. I would say that it is a waste of resources.

I can say that I've never seen software doing this. _MS Word_, for example, shows a list of recent files, rendering only templates. _Diamond_ also shows only a list of files. _Paint.NET_ shows prerendered thumbnails (i.e. even if the file was deleted it still shows them).
I think that the right solution is to show the list of files, showing prerendered thumbnails.

chennes, 2025-04-28:
> I think that the right solution is to show the list of files, showing prerendered thumbnails.

I think maybe I don't understand what you mean by "prerendered thumbnails" -- `f3d` is the tool we're using to *make* those prerendered thumbnails, which we then cache so they are only created once. If you don't have `f3d` installed, then same as in the past you get just an icon for the file type, as determined by your OS. Is your preference to *not* generate the pre-rendered thumbnails, but instead to always only show the icons (for e.g. STEP files)?

luzpaz, 2025-07-27:
bumping for visibility

pixelwaster, 2025-10-07:
FreeCAD 1.1.dev 2025.10.01 no longer uses my installed version which was causing the issues.

closed 2025-10-07 by pixelwaster