FreeCAD/FreeCAD #20979: Array of windows not putting holes in host.

brookermusic, 2025-04-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I make a circular wall, I put a window in it (makes the hole), then I turn that window into an array and the holes for the other windows are not there. Even when I go into the wall and select the array as a subtraction it doesn't work. However if I go into the array and expand it, then change each window's host to the wall it works.

### Full version info

```shell
1.0
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

brookermusic, 2025-04-24:
I should add that this was a polar array.

luzpaz, 2025-04-24:
Please zip up the example file and drag/drop it into the github textbox to attach it to this ticket, TIA. 

We also need your full [About](https://wiki.freecad/org/About) info

Roy-043, 2025-07-10:
This is indeed an issue. But IMO it should be labeled as a feature.