FreeCAD/FreeCAD #20985: BIM: when an object is created out of a base object in a group, ghost base objects appear

furgo16, 2025-04-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Create a BIM file with the customary object hierarchy (Site > Building > Levels) and a group inside one of the levels. You can use the ready-made [Levels_ghosts.zip](https://github.com/user-attachments/files/19904618/Levels_ghosts.zip) file as well. It first needs to be uncompressed.
2. Switch to the BIM workbench
3. Select the group inside the level and create a rectangle base object
4. The base object is created outside of the group => FAIL, but not the subject of this issue
5. Move the base object inside the group it was intended to be in
6. Select the base rectangle object and create a BIM object (e.g. a wall) from it
7. The host object (wall) is created outside of the group => FAIL, but not the subject of this issue
8. Notice there are now two base objects on the tree: one within the host object (wall) and a ghost one inside the group. Moving the host object to be inside the group does not remove the ghost object

https://github.com/user-attachments/assets/836da7db-8624-4d21-a251-0b858e2a6c49

The workaround is to **drag both the ghost base object and the host object to the root of the tree**. Then the ghost object disappears and the host object can be moved back to its original group:

https://github.com/user-attachments/assets/71908bc2-c606-4c66-aaa3-786194576d7e

### Full version info

```shell
1.1.0dev, but the issue has been there since 0.21 at least.
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kaiwas, 2025-04-25:
you can make the group active, but as it often happens this option is hidden. I solve this with scripts)
I will demonstrate later.

also Roy043 made copy tracking. so that copied objects get into the active group.

but I agree that the situation when objects drop out is very inconvenient

semhustej, 2025-04-25:
NOTE: in the workaround, you can drag ONLY the ghost-base object to the top of the tree for it to disappear.

As @kaiwas wrote, you can create objects inside of groups by making the group active in the autogroup . 

[image]

You have to enable groups in the autogroup widget to see them there, by default only layer are visible

[image]

I would prefer to have objects created in selected group in the project tree as you are implying, but this is off-topic.

furgo16, 2025-04-25:
Thanks! Indeed, this is good to know but as we say, unrelated to the issue. I should probably report a separate one for that. In any case, the bug at hand is the appearance of ghost components.

kaiwas, 2025-04-25:
After this important setting (which for some reason is disabled by default), you can select the active group.
But if I understood correctly, the level (if it is made active) will have priority over the group.

Also, the layers (if any) will be visible in the list of active groups.
As a result, everything becomes very complicated. The current behavior is just terrible. Let's create a discussion and discuss the correct behavior.
The main goal is for the object to be created where the user needs it without unnecessary actions.

https://github.com/user-attachments/assets/746b6004-2b5c-4323-bd30-c2b7c2b148b8

kaiwas, 2025-04-25:
The result may be different depending on the choice in the tree view or in the panel with active layers. The tree seems to have priority. It's all very confusing.
The algorithms need to be simplified somehow.

https://github.com/user-attachments/assets/4acb30a8-780d-495e-b74c-778558ee896c

kaiwas, 2025-04-25:
> Thanks! Indeed, this is good to know but as we've said, unrelated to the issue. I should probably report a separate one for that. In any case, the bug at hand is the appearance of ghost components.

This is a confusing program logic. It is impossible to get used to it.
When you dragged a wall into a group, it moved there along with the informant that the wall has a rectangle base.
But the rectangle itself did not move.
The situation is when we have a source rectangle and a rectangle informant.

I know how to quickly fix this problem. Just turn off everything invisible in the tree using a macro and forget about it like a bad dream )))

This is a joke, but still the scene is greatly simplified.

I know how to quickly fix this problem. Just turn off everything invisible in the tree )))

kaiwas, 2025-04-27:
@maxwxyz Hey! You closed my issue without understanding. At least watch the video of what's happening. The topic is completely different. Reopen.
Only the topic title is similar, but this is possible due to translation difficulties.

furgo16, 2025-05-16:
Related forum post: https://forum.freecad.org/viewtopic.php?p=827307#p827307

kaiwas, 2025-05-29:
@furgo16  When the level is active, duplicates of objects are created.
A quick way to eliminate all duplicates is to drag them to the document level.

Thanks to @semhustej semhustej for the tip.
But it would be better if this happened automatically.

https://github.com/user-attachments/assets/aee00139-e1f8-4b71-adc6-94a0816565c4

furgo16, 2025-07-14:
Another related forum post: https://forum.freecad.org/viewtopic.php?p=836503#p836503

semhustej, 2026-08-15:
Still valid.