FreeCAD/FreeCAD #20986: how to disable the rectangular zero-distance auto constrain

mohamdy1997, 2025-04-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

when i draw a rectangular a nineth constrain (constrain distance) which value is **zero** is added automatically even i turned auto constrains off, the rectangular collapses to be a straight line, a message appears **Updating geometry: Error build geometry (0): Both points are equal**
i want to disable this distance auto constrain! pls some help!
this video shows what happens.

https://github.com/user-attachments/assets/43626f8b-9993-494d-8116-c732c5e8b164

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Arabic/Egypt (ar_EG)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-11:
please test if the issue is still persistent in 1.1dev

luzpaz, 2025-05-17:
@mohamdy1997 can you test with 1.1dev[^1] and report back ?

[^1]: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds#assets

FEA-eng, 2025-10-07:
@mohamdy1997 I can't reproduce it. Can you recheck in the newest weekly build?

closed 2025-10-07 by maxwxyz