FreeCAD/FreeCAD #20992: We can't refer to a sketch normal as a subelement but it's used in some properties

FlachyJoe, 2025-04-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When you PartDesign_Pad a sketch, the Reference axis is set to `Sketch [N_Axis]`. However it's not possible to refer this in an expression or to use it as a reference axis to a pad not based on this sketch.

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.41402 +1 (Git)
Build date: 2025/04/24 12:46:37
Build type: Unknown
Branch: fix_20956
Hash: 2e7a6532b87554804bfaa5f6964c17d3a8d712a1
Python 3.13.3, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.16
shiboken 5.15.16, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.8.2.dev
Locale: English/United States (en_US) [ OS: French/France (fr_FR) ]
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.4394/1
Installed mods: 
  * Render 2024.12.15
  * Curves.backup1744720429.3214982 0.6.60 (Disabled)
  * lattice2 1.0.0
  * Lithophane
  * Curves 0.6.61
  * fasteners 0.5.38
  * Assembly4 0.50.18
  * Telemetry 1.0.0
  * Road 2025.2.22
  * FCPDWorkbench 0.4.0
  * sheetmetal 0.7.22
```

### Subproject(s) affected?

Part

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-05:
@FlachyJoe maybe you can give some tips on how someone can implement this feature ?

FlachyJoe, 2025-05-05:
We can add a read-only `Normal` property for sketches so it will be usable in expression. Recompute should be done at placement change.

It would be great to allow a sketch object selection to be automagically transformed in the normal vector when a vector is needed but I don't know if we have the requirement.

maxwxyz, 2026-11-07:
Part of this is implemented in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/7ef5f04245e6053d8662489f892c43bdb098ec62

Sketches get a read-only Normal property that follows Placement and can be used in expressions. It does not add a selectable N_Axis subelement or automatic conversion to a reference axis.