FreeCAD/FreeCAD #20994: Correct fonts for Russian locale drawing template

timursagdenov, 2025-04-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

There are drawing templates for Russian locale in .svg format. They use Arial font. But according to GOST and Unified System for Design Documentation they should use GOST Type A font. I think the font for all the templates for Russian locale should be replaced.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Kazakhstan (ru_KZ)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

leematos, 2025-07-06:
@timursagdenov -- Luzpaz asked me to take a look here and I have some questions:

- Do you have the GOST Type A font installed on your machine?
- Do you know what license this font has? (To consider bundling it, I'd need to be public domain)
- If the editable field had another option to let you choose a typeface: 

[image] would that work?

leematos, 2025-07-06:
In theory if GOST-A is commonly downloaded and on your machine, we should be able to update the font-family here:

https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/TechDraw/Templates/locale/ru/Leading_text/Portrait_A3.svg?short_path=68bec5e#L55-L78

To instead font-family:"GOST-A, Arial, San-Serif" and note in the docs if you want GOST TYPE A to make sure it's installed on your system. (In _theory_ -- lots of unknowns still)

luzpaz, 2025-07-12:
@timursagdenov ping

timursagdenov, 2025-07-14:
>     * Do you have the GOST Type A font installed on your machine?
Yes, I've downloaded it from [here](https://aur.archlinux.org/packages/ttf-gost).
>     * Do you know what license this font has? (To consider bundling it, I'd need to be public domain)
No, sorry.
>     * If the editable field had another option to let you choose a typeface would that work?
It would work, but is there an option to set a default font for Russian templates? It'd be more correct, because literally all drawings I've seen in my work as an engineer (Kazakhstan engineers work following GOST) were made using the GOST fonts.

leematos, 2025-07-19:
@timursagdenov Thanks!

I'm not sure the license BUT I think we can update the CSS to call for GOST Type A. 

The biggest challenge is it _might_ trigger an error like #12199 😬 I'm hoping to do some testing soon 👍

xtemp09, 2025-08-09:
This repository https://github.com/Metrolog/Font.GOST2.304-81/ distributes the fonts under _Apache_ license. Is it suitable for _FreeCAD_?