FreeCAD/FreeCAD #20997: Closing application with multiple changed files shows dialog of a file in the background

semhustej, 2025-04-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

**Feature**

When closing the application with multiple changed files open, the application correctly asks the user to confirm or discard changes made to each file. During this process individual dialogue opens up for each file. In the background, there is one file visible, so the only cue that user has for which file he/she is saving is the name of the file in the dialogue.
See current FreeCAD behavior in the video:

https://github.com/user-attachments/assets/b910ddd2-f9db-44f3-8349-a01a179181f2

**Proposed behavior:**

- It would be helpful for user to have more cues on which file he/she is saving or discarding, if the Save/Discard/Cancel dialogue would also be accompanied by switching to one of the open views of said file.
- Also, it would be nice if the first save/discard question was about the file which is currently active. There is a high probability that user's mind is on this file so the question about that one would be appropriate.

As an example here is how similar situation is handled in BricsCAD (I think it is handled in similar way in Revit):

https://github.com/user-attachments/assets/fc73dadf-7750-4fda-b795-8eb062973e48

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41365 (Git)
Build date: 2025/04/22 11:42:50
Build type: Release
Branch: main
Hash: 84b57d67b50ac64be2246418fb4b49406904f3a9
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.0
```

### Subproject(s) affected?

Core

### Anything else?

This is is not too important, more of a nicety.
It came to me as I saved a wrong file because of being used to BricsCAD/Revit and not reading the dialogue carefully.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

JoaoMartins99, 2025-05-05:
Hello, a colleague ( [@rpronto](https://github.com/rpronto) ) and I are interested in tackling this issue, we would just like to confirm if it is at all still relevant or if anyone has already tried tackling it. Thank you.

xtemp09, 2025-05-05:
@JoaoMartins99, could you review my PR #21135?

pierreporte, 2025-05-05:
Related: #12680

PR #21135 (2025-05-08): Add switching to the document when closing FreeCAD

Closes #20997
commit 54fdd62f (2025-05-08): Gui: Add switching to the document when closing FreeCAD (#21135)

* Add switching to the document when closing FreeCAD

Closes #20997

* Some update of language

* camelCase every variable name used
closed 2025-05-08 by kadet1090