FreeCAD/FreeCAD #21002: Center of mass macro ceased to work in Weekly Build weekly-2025.04.21

gdvranjes, 2025-04-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

when you try to make visible the center of mass in the drawing, by clicking on the eye, the macro freezes. See attached file.

[center_of_mass_bug.zip](https://github.com/user-attachments/files/19912153/center_of_mass_bug.zip)

### Full version info

```shell
OS: Slackware 15.0 x86_64 (KDE//usr/share/xsessions/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41354 (Git) AppImage
Build date: 2025/04/21 18:01:03
Build type: Release
Branch: main
Hash: fd390acd53dae91898145a22d74050d0d648a45b
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: Spanish/Argentina (es_AR)
Stylesheet/Theme/QtStyle: unset/Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/52.445/1
Installed mods: 
  * Plot.backup1736919899.3132463 2024.11.26 (Disabled)
  * CurvedShapes.backup1736919830.8936977 1.0.13 (Disabled)
  * AirPlaneDesign 0.4.1
  * Curves.backup1736919830.987689 0.6.53 (Disabled)
  * OpenDark.backup1725671265.1599011 2024.3.13 (Disabled)
  * CurvedShapes 1.0.13
  * freecad.gears 1.0.0
```

### Subproject(s) affected?

Part

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-04:
Maybe tests should be added to macros ?

luzpaz, 2025-05-17:
@gdvranjes has the macro been addressed yet ?