FreeCAD/FreeCAD #21003: Unable to add "Fillet" to a mirrored element created by a shaped Additive Pipe on a wire

Thovthe, 2025-04-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Saw this asked about on the FreeCAD subreddit here:
https://www.reddit.com/r/FreeCAD/comments/1k26mcd/unable_to_add_fillet_to_a_mirrored_element/

The users were trying to make a fillet between two mirrored Additive Pipe shapes and were unable to do it in freecad even when making a simple copy but were able to export to STEP and use LSPrepost to add the fillet then re-import the step.  Maybe its an OpenCASCADE issue but it equally might not be.  I recommend reading the thread linked above.

### Full version info

```shell
Seems to be FreeCAD 1.0
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-04-27:
Please add an example file and steps how to reproduce. If possible, test if the issue persists on 1.1dev (weekly builds) and provide full about info.
@pieterhijma FYI

ElementW, 2025-04-27:
Reproducible on 1.0.0, 1.1.0dev AppImage, and b14d3a224b89c1ffbbce86ef6e8314518e24ef10.
<details>

```
OS: Void Linux (KDE/wayland)
Architecture: x86_64
Version: 1.0.0.2_voidlinux
Build type: Release
Python 3.13.1, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.7.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * fasteners 0.5.31
  * OpenTheme 2024.9.1
  * FreeCAD-Ribbon 1.8.1.2
  * SearchBar 1.0.2

OS: Void Linux (KDE/xcb)
Architecture: x86_64
Version: 1.1.0dev.41138 (Git) Conda AppImage
Build date: 2025/04/04 17:31:11
Build type: Release
Branch: main
Hash: cab9053be825b8f09d686d5894f333915e4ee378
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * fasteners 0.5.31
  * OpenTheme 2024.9.1
  * FreeCAD-Ribbon 1.8.1.2
  * SearchBar 1.0.2

OS: Void Linux (KDE/wayland)
Architecture: x86_64
Version: 1.1.0dev.41435 (Git)
Build date: 2025/04/26 12:23:25
Build type: Unknown
Branch: main
Hash: b14d3a224b89c1ffbbce86ef6e8314518e24ef10
Python 3.13.2, Qt 6.8.2, Coin 4.0.3, Vtk 9.3.1, boost 1_83, Eigen3 3.4.0, PySide 
shiboken , SMESH 7.7.1.0, xerces-c 3.2.5, OCC 7.7.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * fasteners 0.5.31
  * OpenTheme 2024.9.1
  * FreeCAD-Ribbon 1.8.1.2
  * SearchBar 1.0.2
```

</details>

[mirroredpipe.FCStd.zip](https://github.com/user-attachments/files/19930679/mirroredpipe.FCStd.zip)

Fails with the unclear and unactionable "BRep_API: Command not done" message, see #19255.

Thovthe, 2025-04-28:
Even stranger, some commenters on the [video](https://www.youtube.com/watch?v=GulWjs1WujQ&t=871s) are claiming that exporting the step and reimporting it lets you make the change in FreeCAD.