FreeCAD/FreeCAD #21009: BIM/DRAFT: The object is not created in the active layer.

kaiwas, 2025-04-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

If we make the layer active, the object is created outside of it (we are talking about the tree view). But at the same time the object is painted as expected.

Video demonstration.

https://github.com/user-attachments/assets/6bce9a42-1517-4864-bd57-328988e2047e

For comparison, I create an object on the active level. It is clear that when the level is active, the object is created on the level. But at the same time it is painted as if it belongs to the layer.

That is, something incomprehensible is happening. There is no way to determine in the tree view to which layer the object belongs/

Test file is attached. Remove the zip extension.

[Layer_problem.FCStd.zip](https://github.com/user-attachments/files/19922209/Layer_problem.FCStd.zip)

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41416 (Git) AppImage
Build date: 2025/04/25 15:50:11
Build type: Release
Branch: main
Hash: b2a9f0f7d2cbaf633ecd42f137c3e17871a0fb68
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1
Installed mods: 
  * Reinforcement
  * SvgWorkbench-main 1.0.0.dev5
  * PitchedRoof
  * LyapinWorkbrench
  * FreeCAD_SketchArch-master
  * Road 2025.2.22
  * PitchedRoof.zip
  * TrackingGroups
  * Curves 0.6.61
  * freecad.gears 1.3.0
  * TrackingGroups.zip
  * sheetmetal 0.7.22
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-04-27:
This is a duplicate of #19740.

kaiwas, 2025-04-27:
> This is a duplicate of [#19740](https://github.com/FreeCAD/FreeCAD/issues/19740).

Thank you! If the objects inside which the hidden objects in the tree are located had an identifier, there would be fewer such questions.
I would never have guessed that the problem was in this setting.

So there are two problems here:
- a design error.
- different layer settings depending on the method of their creation.

closed 2025-10-20 by Roy-043