FreeCAD/FreeCAD #21010: BIM/DRAFT: Reorder children for layers does not work.

kaiwas, 2025-04-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

If we create several objects on different layers - we cannot change their order. Although the panel "reorder children" displays the desired result.

Demonstration on video

https://github.com/user-attachments/assets/405d8686-a5fc-49f2-a02c-dd286828f52d

It seems that we do not have a way to change the order of layers. Maybe the manager requires the button "Move selected layer"

[image]

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41416 (Git) AppImage
Build date: 2025/04/25 15:50:11
Build type: Release
Branch: main
Hash: b2a9f0f7d2cbaf633ecd42f137c3e17871a0fb68
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1
Installed mods: 
  * Reinforcement
  * SvgWorkbench-main 1.0.0.dev5
  * PitchedRoof
  * LyapinWorkbrench
  * FreeCAD_SketchArch-master
  * Road 2025.2.22
  * PitchedRoof.zip
  * TrackingGroups
  * Curves 0.6.61
  * freecad.gears 1.3.0
  * TrackingGroups.zip
  * sheetmetal 0.7.22
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-04-27:
Layers are sorted alphabetically in the Group. This overrides the reordering of children.

kaiwas, 2025-04-27:
> Layers are sorted alphabetically in the Group. This overrides the reordering of children.

This should either allow sorting layers without taking into account the alphabet, or make the "Reorder children" option unavailable from the context menu. The first option is better.