FreeCAD/FreeCAD #21012: BIM. Architectural objects do not work correctly if the active group is selected.

kaiwas, 2025-04-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Most do not process the active group correctly if it is selected in the tree. At the same time, for example, lines in DRAFT are created correctly.
In my opinion, groups cannot be the basis for creating objects. And this is exactly how a group behaves. This is fundamentally wrong.

Video demonstration. I start creating a line in DRAFT (I get the expected result).

https://github.com/user-attachments/assets/5bc0e001-5af1-4a7e-b755-67b8a52a297b

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41416 (Git) AppImage
Build date: 2025/04/25 15:50:11
Build type: Release
Branch: main
Hash: b2a9f0f7d2cbaf633ecd42f137c3e17871a0fb68
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1
Installed mods: 
  * Reinforcement
  * SvgWorkbench-main 1.0.0.dev5
  * PitchedRoof
  * LyapinWorkbrench
  * FreeCAD_SketchArch-master
  * Road 2025.2.22
  * PitchedRoof.zip
  * TrackingGroups
  * Curves 0.6.61
  * freecad.gears 1.3.0
  * TrackingGroups.zip
  * sheetmetal 0.7.22
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

BIM

### Anything else?

I plan to create a discussion on the topic of active layers. I think that if I make a few simple (in my opinion) changes, the situation with placing objects in the tree will be much more predictable. The current situation is simply terrible and very confusing. I am gathering my thoughts and analyzing the parts that will need to be corrected. There are a lot of problems, but they can be solved in just a couple of simple steps.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-04-27:
I think it is the same issue as https://github.com/FreeCAD/FreeCAD/issues/20985

kaiwas, 2025-04-27:
> I think it is the same issue as [#20985](https://github.com/FreeCAD/FreeCAD/issues/20985)

No!  At least watch the video of what's happening.
Only the topic title is similar, but this is possible due to translation difficulties.

maxwxyz, 2025-04-27:
So is the issue that the new object is created and leaves the base object still next to it inside the group (at least this is what I am seeing in the video)? Or is the issue that it should not be possible to create new BIM objects with a group as base feature?

kaiwas, 2025-04-27:
> Or is the issue that it should not be possible to create new BIM objects with a group as base feature?

Yes. A group cannot be a base.
Any group object (group/building/level/layer) is "virtual".

It also does not generate any geometry.

semhustej, 2025-04-27:
I quite often run into similar problems, especially creating Window object when other window is selected -> new window is create on a basis of the first window object and creates incorrect result.

Would it make sense to create a list of BIM commands, which have a base object on which they are created, where this creates unwanted results, such as the window with a window as a base or wall with group as a base? If such a list existed, developers could tackle these cases one by one.

So the list would look like:

**Base object  -  BIM command**
Group - BIM_Wall
Arch_Window - Arch_Window

kaiwas, 2025-04-27:
> I quite often run into similar problems, especially creating Window object when other window is selected -> new window is create on a basis of the first window object and creates incorrect result.

Yes. I also think that some filter is needed.
But nevertheless - the ability to make a "wrapper" is quite important. I would not like to lose it.

For example, there is nothing wrong with wrapping a column with a wall. After all, a column can be extended and essentially be a wall.

Or when using the pitched roof add-on, a mesh is generated, which is then extruded along the Z axis. Then it is wrapped with a roof.

[image]

furgo16, 2025-04-27:
> it should not be possible to create new BIM objects with a group as base feature

I would suggest changing the issue title to the above sentence, for clarity.

kaiwas, 2025-04-27:
> I would suggest changing the issue title to the above sentence, for clarity.

I don't mind. Write a specific phrase from the title and I'll change it (I'm afraid to make a mistake with the translation)