FreeCAD/FreeCAD #21013: Crash while using sketcher trim tool

lagnat, 2025-04-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I've experienced this crash with the weekly builds going back several months.

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [81621]
Path:                  /Applications/FreeCAD.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-04-26 09:58:03.6419 -0400
OS Version:            macOS 15.4.1 (24E263)
Report Version:        12
Anonymous UUID:        80057660-6D13-007C-563C-EE545423E9F4

Sleep/Wake UUID:       79D601A4-640F-4F28-82CF-C0F83CB04B1F

Time Awake Since Boot: 430000 seconds
Time Since Wake:       60 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x000058eda4708f80
Exception Codes:       0x0000000000000001, 0x000058eda4708f80

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [81621]

VM Region Info: 0x58eda4708f80 is not in any region.  Bytes after previous region: 97296652996481  Bytes before following region: 7775426932864
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      commpage (reserved)        1000000000-7000000000   [384.0G] ---/--- SM=NUL  reserved VM address space (unallocated)
--->  GAP OF 0x5f9000000000 BYTES
      MALLOC_NANO              600000000000-600020000000 [512.0M] rw-/rwx SM=PRV  

Kernel Triage:
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   Sketcher.so                   	       0x16e2f32f4 Sketcher::SketchObject::trim(int, Base::Vector3<double> const&) + 1208
1   Sketcher.so                   	       0x16e3fc9e8 Sketcher::SketchObjectPy::trim(_object*) + 136
2   Sketcher.so                   	       0x16e3ecb78 Sketcher::SketchObjectPy::staticCallback_trim(_object*, _object*) + 116
3   libpython3.11.dylib           	       0x102d7e6f4 cfunction_call + 284
4   libpython3.11.dylib           	       0x102d26968 _PyObject_MakeTpCall + 332
5   libpython3.11.dylib           	       0x102e2a064 _PyEval_EvalFrameDefault + 45376
6   libpython3.11.dylib           	       0x102e1df48 PyEval_EvalCode + 204
7   libpython3.11.dylib           	       0x102e8455c run_mod + 144
8   libpython3.11.dylib           	       0x102e8824c PyRun_StringFlags + 112
9   libFreeCADBase.dylib          	       0x102a1c958 Base::InterpreterSingleton::runString(char const*) + 108
10  libFreeCADGui.dylib           	       0x104be0624 Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) + 168
11  libFreeCADGui.dylib           	       0x104be04b8 Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) + 120
12  SketcherGui.so                	       0x17ecba1ec void Gui::cmdAppObjectArgs<int&, double&, double&>(App::DocumentObject const*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, int&, double&, double&) + 280
13  SketcherGui.so                	       0x17ecb9fe4 SketcherGui::DrawSketchHandlerTrimming::executeCommands(Base::Vector2d) + 324
14  SketcherGui.so                	       0x17ecb9b10 SketcherGui::DrawSketchHandlerTrimming::releaseButton(Base::Vector2d) + 16
15  SketcherGui.so                	       0x17edb5094 SketcherGui::ViewProviderSketch::mouseButtonPressed(int, bool, SbVec2s const&, Gui::View3DInventorViewer const*) + 1920
16  libFreeCADGui.dylib           	       0x104f73f38 Gui::ViewProvider::eventCallback(void*, SoEventCallback*) + 544
[... 59 lines cut ...]
Version: 1.1.0dev.41416 (Git)
Build date: 2025/04/25 15:50:11
Build type: Release
Branch: main
Hash: b2a9f0f7d2cbaf633ecd42f137c3e17871a0fb68
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/macos
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * freecad.gears 1.3.0
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

JohnOCFII, 2025-04-26:
Confirmed with both the current weekly and the new main-repository builds weeklies.

To recreate:

- Enter Sketcher (via PartDesign in my testing)
- Create Sketch (XY plane in my testing)
- Create Rectangle (without dimensions)
- Create intersection line as shown
- Select Trim tool
- Select line segment near upper left corner (as shown in attached screen shot)
- ** BOOM! **

[image]

```
New Builds

OS: macOS 15.4.1
Architecture: arm64
Version: 1.1.0dev.41354 (Git)
Build date: 2025/04/21 18:01:03
Build type: Release
Branch: main
Hash: fd390acd53dae91898145a22d74050d0d648a45b
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * OpenTheme 2024.9.1
  * Telemetry 1.0.1

Legacy Builds

OS: macOS 15.4.1
Architecture: arm64
Version: 1.1.0dev.41416 (Git)
Build date: 2025/04/25 15:50:11
Build type: Release
Branch: main
Hash: b2a9f0f7d2cbaf633ecd42f137c3e17871a0fb68
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * OpenTheme 2024.9.1
  * Telemetry 1.0.1
```

Crash Log (both similar, so only loading one):

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [7744]
Path:                  /Applications/Other Apps/FreeCAD-41416.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-04-26 10:49:02.1031 -0500
OS Version:            macOS 15.4.1 (24E263)
Report Version:        12
[... 74 lines cut ...]
46  libQt6Widgets.6.7.3.dylib     	       0x106fa0004 QApplication::notify(QObject*, QEvent*) + 508
47  libFreeCADGui.dylib           	       0x1049ac178 Gui::GUIApplication::notify(QObject*, QEvent*) + 284
48  libQt6Core.6.7.3.dylib        	       0x107d9ad50 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 212
49  libQt6Gui.6.7.3.dylib         	       0x10768d40c QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) + 1808
50  libQt6Gui.6.7.3.dylib         	       0x10760df14 QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 408
51  libqcocoa.dylib               	       0x1096fdb20 QCocoaEventDispatcherPrivate::postedEventsSourceCallback(void*) + 436
52  CoreFoundation                	       0x18305cde4 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 28
53  CoreFoundation                	       0x18305cd78 __CFRunLoopDoSource0 + 172
54  CoreFoundation                	       0x18305cae4 __CFRunLoopDoSources0 + 232
55  CoreFoundation                	       0x18305b738 __CFRunLoopRun + 840
56  CoreFoundation                	       0x18305ad68 CFRunLoopRunSpecific + 572
57  HIToolbox                     	       0x18e80e27c RunCurrentEventLoopInMode + 324
58  HIToolbox                     	       0x18e81131c ReceiveNextEventCommon + 216
59  HIToolbox                     	       0x18e99c484 _BlockUntilNextEventMatchingListInModeWithFilter + 76
60  AppKit                        	       0x186cadab4 _DPSNextEvent + 684
61  AppKit                        	       0x18764c314 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 688
62  AppKit                        	       0x186ca0c64 -[NSApplication run] + 480
63  libqcocoa.dylib               	       0x1096fb70c QCocoaEventDispatcher::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 2452
64  libQt6Core.6.7.3.dylib        	       0x107d9bb48 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 576
65  libQt6Core.6.7.3.dylib        	       0x107d9b78c QCoreApplication::exec() + 128
66  libFreeCADGui.dylib           	       0x1048e36ac Gui::Application::runApplication() + 2420
67  freecad                       	       0x10229ce28 main + 4244
68  dyld                          	       0x182bd2b4c start + 6000

```

maxwxyz, 2025-04-27:
@AjinkyaDahale FYI

AjinkyaDahale, 2025-04-27:
Could someone check post compiling https://github.com/FreeCAD/FreeCAD/pull/20951?

JohnOCFII, 2025-04-27:
> Could someone check post compiling [#20951](https://github.com/FreeCAD/FreeCAD/pull/20951)?

Since it was only a few lines changed, I hand edited my local copy of `src/Mod/Sketcher/App/SketchObject.cpp` and your recent changes resolves this issue.  Hopefully it can get merged soon.

[image]

maxwxyz, 2025-07-21:
so is this resolved?

JohnOCFII, 2025-07-21:
> so is this resolved?

It works for me (as shown above, and just retested in current weekly).

```
OS: macOS 15.5
Architecture: arm64
Version: 1.1.0dev.42605 (Git)
Build date: 2025/07/18 16:20:04
Build type: Release
Branch: main
Hash: da5e18f5d8e7eb4df89c86c9562599435b5c5187
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * OpenTheme 2025.5.20
  * AddonManager 2025.7.17
  * Telemetry 1.0.1
```

[image]