FreeCAD/FreeCAD #21014: Hovertext freezes in freecad

StenellaCoeruleoalba, 2025-04-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hovering an icon long enough for hovertext to appear causes Freecad to freeze
Here are the videos of the error
https://upload.disroot.org/r/CenRfVPp#JtaMjoJiBtG8r62k/EcstP/m0eXXLJj7SQWSbwFPdvM=
https://upload.disroot.org/r/re76KraB#sSPn2yVHTAQDwJTUL5WFJnyQJ/IkR632QjuIYm/2McU=

### Full version info

```shell
https://upload.disroot.org/r/CenRfVPp#JtaMjoJiBtG8r62k/EcstP/m0eXXLJj7SQWSbwFPdvM=
https://upload.disroot.org/r/re76KraB#sSPn2yVHTAQDwJTUL5WFJnyQJ/IkR632QjuIYm/2McU=
Now, when using 1.1 whenerver I hover an icon with my mouse long enough for the text description to happen, Freecad freezes
I'm not even sure what this sort of bug is called
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

JohnOCFII, 2025-04-26:
Please include full version info from the "About FreeCAD menu item"

StenellaCoeruleoalba, 2025-04-26:
[image]

JohnOCFII, 2025-04-26:
> Turns out that copying the version info to the clipboard causes it to freeze too...

@StenellaCoeruleoalba It doesn't completely freeze.  Give it 20 seconds... (Pretty sure I already reported this -- it is not consistent across operating systems/themes.)

StenellaCoeruleoalba, 2025-04-26:
> > Turns out that copying the version info to the clipboard causes it to freeze too...
> 
> [@StenellaCoeruleoalba](https://github.com/StenellaCoeruleoalba) It doesn't completely freeze. Give it 20 seconds... (Pretty sure I already reported this -- it is not consistent across operating systems/themes.)

I will try again

StenellaCoeruleoalba, 2025-04-26:
[Freezelog.rtf.zip](https://github.com/user-attachments/files/19924903/Freezelog.rtf.zip)

maxwxyz, 2025-04-27:
Can you please try again in safe mode (Help>Restart in safe mode). Thanks!

StenellaCoeruleoalba, 2025-04-27:
> Can you please try again in safe mode (Help>Restart in safe mode). Thanks! Does it also happen in the stable version 1.0 on your system?

It's specific to the 1.1 dev branch, I haven't had any of this with 1.0

StenellaCoeruleoalba, 2025-04-27:
Safe mode does it too

luzpaz, 2025-05-04:
@StenellaCoeruleoalba JFYI, next time please encode the videos to a lower quality so you can directly add them to the ticket instead of a 3rd party site that may eventually break. I've re-encoded the clips and updated the OP. TIA.

maxwxyz, 2025-05-05:
https://github.com/FreeCAD/FreeCAD/issues/21094

closed 2025-05-05 by maxwxyz