FreeCAD/FreeCAD #21017: Part Design Hole tool: no longer possible to make a hole from one selected circle in a sketch

depthoffocus, 2025-04-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Since the new Hole functionality for arcs and points has been added, it seems to be impossible to select a single circle from a sketch (i.e. not all circles) to make a hole from?

https://github.com/user-attachments/assets/9dafe30b-ebf2-4de4-8de1-59e7d7148861

This is possible in 1.0 and was possible in the previous weekly I tried (build 41354):

https://github.com/user-attachments/assets/991318e3-0630-4ccc-8bbb-4f129e79690a

### Full version info

```shell
OS: macOS 15.4.1
Architecture: x86_64
Version: 1.1.0dev.41416 (Git)
Build date: 2025/04/25 15:50:11
Build type: Release
Branch: main
Hash: b2a9f0f7d2cbaf633ecd42f137c3e17871a0fb68
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.75/2
Installed mods: 
  * ThreadProfile 1.94.0
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.38
  * lattice2 1.0.0
  * CurvedShapes 1.0.13
  * sheetmetal 0.7.22
  * Curves 0.6.61
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

JohnOCFII, 2025-04-26:
Confirmed.

[image]

[HoleTest.FCStd.zip](https://github.com/user-attachments/files/19924524/HoleTest.FCStd.zip)

```

OS: macOS 15.4.1
Architecture: arm64
Version: 1.1.0dev.41416 (Git)
Build date: 2025/04/25 15:50:11
Build type: Release
Branch: main
Hash: b2a9f0f7d2cbaf633ecd42f137c3e17871a0fb68
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * OpenTheme 2024.9.1
  * Telemetry 1.0.1
```

depthoffocus, 2025-04-26:
You also can't drop a hole into a single face -- say if you extrude the circles sketch as pegs and then pick only one of them to put holes in, you will get holes in both.

kaiwas, 2025-04-26:
@depthoffocus videos are not playing.

pierreporte, 2025-04-26:
@theo-vt could it be related to your PR #20583?

depthoffocus, 2025-04-27:
> [@depthoffocus](https://github.com/depthoffocus) videos are not playing.

They work everywhere I have been able to test them. But at any rate it's only a very simple project:  place a sketch with two circles on a face of any object, and then try to select just one of the circles in the sketch to make a Hole.

Here's an even stranger outcome. Three separate standoff pads at different heights from three separate sketches. Select a face of any one of them for a hole, and they all get holes, which is also a change from 1.0 and 41354. 

(The behaviour in this case is not the same as if you select the Pad at the tip and then select the Hole tool.)

https://github.com/user-attachments/assets/31da308b-ea4c-46fc-ab2d-24047cb88eef

theo-vt, 2025-04-27:
> [@theo-vt](https://github.com/theo-vt) could it be related to your PR [#20583](https://github.com/FreeCAD/FreeCAD/pull/20583)?

It would seem so, looking into it!

theo-vt, 2025-04-27:
In a case like this should I make a second PR solving this or is there a way to amend #20583 ?

pierreporte, 2025-04-27:
@theo-vt You need to make a second PR.

PR #21030 (2025-05-08): PartDesign: Fix no longer possible to make a hole from one selected circle in a sketch
Add a parameter to prevent simple profiles from getting simplified while keeping sub-element search when the profile is extracted to find hole centers. 

## Issues

fixes #21017 which was introduced in PR #20583
closed 2025-05-08 by kadet1090