FreeCAD/FreeCAD #21021: BIM/UX: Improve the logic of building architectural models (document structuring issue). Simple steps for boost 1000% for convenience.

kaiwas, 2025-04-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

This problem mainly concerns only the BIM workbench. Although some of this logic could be useful for the PART and DRAFT workbenches (maybe somewhere else).
If we briefly describe the problem, we can say this: "It is very difficult to maintain the document structure now." The current version of the document structure is more suitable for the linear construction of a tree (mechanics). The tree view is built from top to bottom and to maintain the structure we have to either drag objects in the tree or click on many different settings and monitor them (and this often leads to errors and is inconvenient). At the same time, the concept of mechanics does not suit us, because each object can be autonomous (and not the result of the general construction). Correct cataloging is much more important.

To place an object in the right place, we have a working plane, an active level (if the double-click setting is activated), an active group, an active layer, and selecting an object in the tree view. It is very difficult to monitor all of this. We need a simpler and more stable option. Everything is already in the program and we only need to slightly change the interaction logic.

For example, we can place an object inside a group. To do this, we must select the active group in the panel. As we can see, there is no synchronization between the tree view and the panel with the active group if we did not have any objects selected before the selection. But synchronization occurs if a group was selected in the tree view and we press the button on the panel.
If at this moment we have an active group selected in the tree view, then the object wraps the group and an error occurs.  https://github.com/FreeCAD/FreeCAD/issues/21012
If we do not have a group selection, then the object falls down the tree.

https://github.com/user-attachments/assets/5a665e04-e7ad-4030-9e4c-41a71d8ba60f

It is very difficult to follow this whole order. We constantly had to think about what we have selected at the moment and what result it will lead to. There is a great chance of making a mistake.

These inconveniences also arise with objects of the "Level/building/layers" type.

-----------------------------------------------------
My suggestion to eliminate these inconveniences. They are quite simple in meaning. I hope that they are also simple in implementation.
1) First, we need to fix this problem https://github.com/FreeCAD/FreeCAD/issues/21012
2) Make constant synchronization between the panel with active groups and the tree view. The concept is simple. If a group is selected in the tree (with a single click), then this group immediately becomes active in the panel. And vice versa, if an active group is selected, it is selected in the tree view. 
If we click on an empty space in the 3D window - we reset the active group. Return to point A. In this case, newly created objects will fall down the tree (the current default version of the program)

The concept is in the picture.

[image]

The picture shows synchronization using a group. But it should work with a level and a building and layers.
The general principle is that what is selected is active!

3) After implementing point 2, we can delete the property for floors and a building that is located in an inconvenient place (VIEW tab)

[image]

4) The BIM panel should also work with synchronization.
Double clicking on a level should select the level in the tree view and set this level active.

[image]

All this will allow you to quickly place objects in the right place and conveniently structure the document.

Perhaps I missed some small features, but I think that the general concept will simplify the construction of architectural objects.

@yorikvanhavre @Roy-043 @BIM @FreeCAD/design-working-group  Your opinion is quite important

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41416 (Git) AppImage
Build date: 2025/04/25 15:50:11
Build type: Release
Branch: main
Hash: b2a9f0f7d2cbaf633ecd42f137c3e17871a0fb68
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1
Installed mods: 
  * Reinforcement
  * SvgWorkbench-main 1.0.0.dev5
  * PitchedRoof
  * LyapinWorkbrench
  * FreeCAD_SketchArch-master
  * Road 2025.2.22
  * PitchedRoof.zip
  * TrackingGroups
  * Curves 0.6.61
  * freecad.gears 1.3.0
  * TrackingGroups.zip
  * sheetmetal 0.7.22
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-04-27:
@FreeCAD/cad-working-group FYI

kaiwas, 2025-04-28:
If groups could be activated by double-clicking (like levels and buildings) - the problem would be solved in general.
It looks weird in the tree view, but essentially solves the problem.

I can always activate the desired group and not even pay attention to the panel with active groups.
A workaround is to use buildings as a group. Activate by double-clicking.
Demonstration on video.

https://github.com/user-attachments/assets/e32be77c-66d5-4372-afd3-8cb4c0c10a0a

It turns out that it is enough to activate the group by double-clicking and the problem will be solved. (the setting activate by double-clicking should be enabled by default)

kaiwas, 2025-05-02:
> This made me think about ordered and unordered containers. Are you also looking for this? In ordered containers, you can only select children in lines above the active one, while unordered container allows you to do anything. In catia, geometric sets work like this, while bodies are ordered. But the document object (part or assembly) isn’t and you can reorder their children freely.

The problem of cataloguing (structuring) is very important. But achieving correct behavior in freecad is almost impossible (at the moment). This is very sad.

furgo16, 2025-05-02:
Thanks for taking the time to think about this and to describe your thoughts in detail.

## Summary

This is my understanding of your proposal. Please correct me if I did not properly understand it.

1. https://github.com/FreeCAD/FreeCAD/issues/21012
1. In the Autogroup dropdown, automatically choose the group that is selected in the tree view
1. Remove the `DoubleClickActivates` property for containers (e.g. Groups, BuildingPart derivatives such as Arch_Floor, etc)
1. In the Autogroup dropdown, automatically choose the group that is selected in the BIM Views panel 

## Feedback

In general, I think these points make sense, but I don't think fixing them would address the more holistic workflow streamlining that I think you're hinting at. To me it feels like a list of issues that improve one small aspect of the workflow.

Feedback to each particular point:

1. This is important, but I don't see how it is related to improving the workflow
2. This should probably be filed as a standalone issue
3. If `DoubleClickActivates` is removed, are you suggesting that whenever a container is selected in the tree view its workplane is automatically selected? I think this automatic workplane switching would come unexpected to users
4. Same comment as 2.

kaiwas, 2025-05-02:
At the moment, I almost managed to get a workflow that suits me. But as a workaround, I am forced to use a "building" instead of groups and a macro to quickly enable double-click activation. Another inconvenience is that the z-coordinate of the building-group also has to be changed.
Quick demonstration on video

https://github.com/user-attachments/assets/a11e12b3-85ff-4e7a-a8f1-46c193cd0fc7

kaiwas, 2025-05-02:
> See behavior:

Keeping track of what you have selected (or not selected) in the tree is a nuisance. It's easy to get confused.
There must be an easier way.
For now, double-clicking to activate works best.