FreeCAD/FreeCAD #21024: BIM Views Panel - ebable selection of multiple objects

semhustej, 2025-04-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

**Feature**

**Current state**
Currently, BIM Views Panel only enables users to select one objects at a time. This limits users to work with one object at a time.

**Proposed solution**
Enable selection of multiple objects in BIM Views Panel using standard `Ctrl` or `Shift` + LMB clicking method. The same way it works in the Project tree. This would enable user to RMB click and select an action from the context menu, which would be applied to all selected objects.

[image]

NOTE: it is currently possible to select multiple objects in project tree or 3D view and apply functions from the BIM Views Panel. See current behavior in the video:

https://github.com/user-attachments/assets/d46abee5-c5e6-4d9a-b3d6-ca0b7db33889

 

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41365 (Git)
Build date: 2025/04/22 11:42:50
Build type: Release
Branch: main
Hash: 84b57d67b50ac64be2246418fb4b49406904f3a9
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.0
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tetektoza, 2025-04-27:
Yes, I also noticed this, but to make it work properly imo BIM Views requires a bit of refactor. It doesn't work properly now, because we basically call update method in a timer, which in turn cleans the whole list and repopulates it once again with new items, so if I guess correctly - this also cleans up the selection. This is also why if you select in "Model" list, you see some sort of a delay for that item to be selected in BIM Views list (it has been set to 2 seconds). What's even worse, is that the View Tree is actually synchronized with the global area, not with the Model Tree. What it means, is it's not aware which item in Model Tree has been selected, it selects it's own items in the list based on the currently selected object in viewport (which Model View Tree selects), so it's aware *what* Model View Tree selected by checking viewport, but it's not aware what it selected by checking Model View Tree itself. And why it would be useful if the tree would be aware of the Model View Tree? Because then we could remove the timer altogether and just update the tree whenever a change happened in Model View Tree (i.e object has been deleted, edited, added).

**Instead of making some wonky changes, I would remove that timer at all and just sync the tree with Model View Tree and I think it's the only acceptable approach here instead of extending current idea and making it more convoluted.**

I've checked it, and managed to sync our Model Tree View with the Bim Tree View, so now it's basically populated everytime we see a change, BUT I need some time to work on it, because it looks like `update` modifies the objects somehow, which causes another `update` calls to be fired, which in turn gets us in the loop. Adding some signal blockers do not help here as it seems some parts are not being initialized properly for some reason.

Gotta investigate it more.

Current progress:

https://github.com/user-attachments/assets/9d89d072-9bf3-45f2-97d5-016fceec3666

But it still requires a bit of work.

semhustej, 2025-04-27:
Thanks for looking into the issue.
I can't really tell what was the main thought about the timer and I didn't know it checks only the 3D view.
From a user perspective it's a bit weird, as it behaves a with a little bit of a lag, but that is mostly not a problem.

semhustej, 2025-05-07:
NOTE: Need to also test this scenario:

`unlike the tree view, you have to always first click on an item on the BIM Views panel to act on it. This leads to confusing behavior whereby hovering an item would appear to select it (it is highlighted)`