FreeCAD/FreeCAD #21028: Rendering issue - Dell driver i915 + Intel Core2 Duo E8400 - Linux Mint 21.3 Cinnamon

Groberen, 2025-04-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

On this particular system, any graphics in any workbench (static graphics such as toolbars, tree-view and bottom bar not affected).

The computer are a Dell OptiPlex 780 (2013).

The issue: All rendering on screen are very pixelated/jagged.
Running FreeCAD in safe mode does not change this behavior.

I've tried to change rendering settings, but doesn't make it better - Change "Anti-Aliasing" to "Line smoothing" make it way worse, as any transparent or semi transparent objects then get completely filled with some color, so bad that it's get impossible to work with this setting activated.

Sample image on the pixelated view:
[image]

### Full version info

```shell
OS: KDE Flatpak runtime (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/

Also, this appimage are tested on same computer, having same rendering issue:
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.41264 (Git) Conda AppImage
Build date: 2025/04/14 16:28:55
Build type: Release
Branch: main
Hash: 31b0dbaa79da9e0a8d18c009006a2b1c6cd16404
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1
```

### Subproject(s) affected?

None

### Anything else?

forum thread for same issue.
https://forum.freecad.org/viewtopic.php?t=96602

**Settings I've already tested**
Settings --> Display 3D View --> Rendering

Use software OpenGL
Initial state: OFF
Result when turning on & restart FreeCAD : No change

Use OpenGL VBO . . .
Initial state: ON
Result when turning off & restart FreeCAD : No change

Anti-Aliasing
Initial state: None
Result after changed to "Line smoothing" & restart FreeCAD : Initially no visible change, but the x-y-z plane selector are completely wasted (replaced by a solid block of blue color) - so FreeCAD cannot be used unless this setting are "None".

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-07:
Can you reproduce on appimage ?

xtemp09, 2025-05-07:
@Groberen, have you tried _Mesa_?

Groberen, 2025-05-07:
@luzpaz Yes I can reproduce on appimage.

@xtemp09 Seems that the Mesa version used in this Linux Mint version is

```
$ glxinfo | grep "OpenGL renderer"
OpenGL renderer string: Mesa Intel(R) Q45/Q43 (ELK)
```

Groberen, 2025-05-07:
Update: I did seek some help on Linux Mint forum on this issue (about the question if tried Mesa).
forum url: https://forums.linuxmint.com/viewtopic.php?t=445686&sid=e3aab46256c7a254342320e14aefadec

What other post in same thread - in short:

- Mesa is a standard Linux program, also included in Linux Mint.
- Mesa cannot be downgraded in Linux (testing older version is out of the question).

FEA-eng, 2025-10-07:
@Groberen Can you confirm if this issue is still relevant?

closed 2026-03-14 by luzpaz