FreeCAD/FreeCAD #21031: UX: Expression editor should not need to be its own modal window

ElementW, 2025-04-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

As it stands, the expression editor only being available as a modal window has multiple drawbacks:

* Modal dialogs are interruptive by design, but inserting/editing a formula is part of a normal flow and doesn't justify this. This forces users to move their focus away from fields that already pertain to the metric they wanted to edit, and then back.
* It is currently devoid of context as to what you are editing, not carrying a property name or, crucially, a type of unit (e.g. distance, angle, ...), leading to #16861
* Most expressions are short enough to reasonably fit (± some scrolling or vertical expansion) within the fields they affect the value of
* It blocks focus on the main window for no valid reason
  * Will have to be solved for the implementation of #12146
* It can, under various circumstances, spawn partially or completely off-screen: #12591
  * Multi-monitor setups suffer from this: #18119
  * Relatedly, its expansion mechanism causes it to spill off the screen bounds: #12518
* It is regularly *a modal of a dialog*, for example as a formula for the "Insert angle" dialog of `Sketcher_ConstrainAngle`; this is considered bad form
* (Should be its own issue) Its "Clear" and "OK" buttons are ordered and positioned in total opposition to the established modal window button model.

Inline expression editing should be available in most if not all places where an expression can be used (that is, everywhere a `QuantitySpinBox` or numeric `PropertyItem` is used [1]) and provide the same editing facilities the modal window does. This would provide a smoother editing experience for a lot of use cases, while keeping the possibility of opening a bigger, possibly more complete editor.

[1]: For the latter, if keyboard navigation of the property editor is planned, #15065 should be taken in account.

### Full version info

```shell
Any FreeCAD version up to and including 47c1565edfce18f3aeb87e8d8f436dda473d174f
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

pierreporte, 2025-04-28:
I agree. Maybe a modal window could be useful if it had more options, but hitting <kbd>=</kbd> like in spreadsheet programs shouldn’t.

Thovthe, 2025-04-30:
Absolutely third this.  It would be nice if it could even be done without the `=` syntax.  I think there are a few other improvements that could be made to the UX of expressions but making it non-modal would really make using expressions a lot smoother by itself.

julianjc84, 2025-06-07:
Some UI features on how the Expression editor presents the (List of suggested candidates) behaves differently for <<labels>> vs internal names.
https://forum.freecad.org/viewtopic.php?t=97548