FreeCAD/FreeCAD #21032: BIM: Add offset parameter to wall parameters

kaiwas, 2025-04-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Currently, the offset property is only available through the properties panel. But this property is quite important when building multi-layer walls (without multi-materials).

[image]

You can add the parameter to the settings of the wall being created.
When the alignment is center - offset cannot be set.
When the alignment is left or right - offset is a configurable parameter.

[image]

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41416 (Git) AppImage
Build date: 2025/04/25 15:50:11
Build type: Release
Branch: main
Hash: b2a9f0f7d2cbaf633ecd42f137c3e17871a0fb68
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * addFC
  * OpenTheme 2024.9.1
  * PitchedRoof
  * Test
  * sheetmetal 0.7.22
  * Road 2025.2.22
  * Reinforcement
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kaiwas, 2025-04-28:
> Could you elaborate?

I'm not sure, but maybe always make the value 0 if centered?
However, if the value is not 0, but does nothing, this won't be a problem.

furgo16, 2025-04-28:
> However, if the value is not 0, but does nothing, this won't be a problem.

That's how currently it is implemented in the PR.

However, I'd like to know if there is a specific reason or use case to disable the offset input if the alignment is Center. Otherwise, it does not make sense to disable it.

kaiwas, 2025-04-28:
I don't think it affects anything

furgo16, 2025-04-28:
Issue accidentally closed?

kaiwas, 2025-04-28:
> Issue accidentally closed?

Yes. Sorry. I didn't notice how I did it. Maybe the small screen of the smartphone is to blame.

furgo16, 2025-04-28:
I've found out that on the Property Editor, with an offset defined but Align set to Center, the offset value is ignored. I do not know why, but the PR now follows the behaviour outlined in this issue description. That is, entering an offset in the Task panel when creating a new wall is only possible if the alignment is either Left or Right.

In any case, the PR is ready. I'm leaving it as a draft for now, as it seems to output a warning that I don't understand where it comes. Otherwise the functionality seems not to be affected by the warning.

kaiwas, 2025-04-29:
> I've found out that on the Property Editor, with an offset defined but Align set to Center, the offset value is ignored.

This is in line with the properties panel capabilities.

we can change the offset for any alignment.
But for center alignment it has no effect (and is not reset)

[image]

paullee0, 2025-05-01:
> I've found out that on the Property Editor, with an offset defined but Align set to Center, the offset value is ignored. I do not know why, but the PR now follows the behaviour outlined in this [...]

With Centre set, the user intend to offset from which direction is not sure - to the 'Left' or 'Right' direction.

I had thought about creating something like CentreL, CentreR, but it seems over-complicated and confusing.

closed 2025-05-19 by hyarion