FreeCAD/FreeCAD #21034: BIM: Add the ability to select a base for a wall/beam/column object from the taskbar.

kaiwas, 2025-04-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The translation may not be correct (correct if necessary). I will briefly describe the essence.

At the moment, we have two orders of actions for creating a wall object.
1) We created a base and clicked on the tool.
2) We clicked on the tool and set the wall by moving the cursor.

I think it would be useful to be able to select a base at the initial stage of point 2. That is, when the taskbar already has settings (by the way, we can change them if necessary), but we have not yet moved the cursor and set the first and second points.

In short, we "apply" the wall tool when it is open by selecting a base at the moment when the wall settings are displayed on the taskbar.

Approximate concept in the video.

https://github.com/user-attachments/assets/43b2a525-6cea-45b1-ab88-bc20b2aae8d4

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41416 (Git) AppImage
Build date: 2025/04/25 15:50:11
Build type: Release
Branch: main
Hash: b2a9f0f7d2cbaf633ecd42f137c3e17871a0fb68
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * addFC
  * OpenTheme 2024.9.1
  * PitchedRoof
  * Test
  * sheetmetal 0.7.22
  * Road 2025.2.22
  * Reinforcement
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-04-28:
Would you mind changing "taskbar" to "Task panel"? https://wiki.freecad.org/Task_panel. The rest is clear. Thanks!

furgo16, 2025-04-28:
So the workflow you're proposing is:

1. Click on the `Arch_Wall` command on the toolbar
2. When the Task panel is shown to enter the first point, optionally change settings (height, width, alignment)
3. Have some means to pick a base object from the Task panel
4. Close Task dialog. The wall is created using the base object and the defined settings

Do I get this right?

Do you have a specific proposal on how to pick the base object in step 3?

markkomartin, 2025-04-28:
> The general concept is: "Activate the tool. Change the settings if necessary (this is the main thing). Apply to the selected

I agree with your suggestion. It would be a good update when importing a 2D drawing as a base. 
Is it currently possible to display the wall options task panel of an existing wall? 
In my opinion it could be useful to have it both during the creation of the wall and also after having created it.

[image]