FreeCAD/FreeCAD #21038: PATH CommandPy::setParameters isn't a setter.

J-Dunn, 2025-04-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

in CommandPyImp.cpp  CommandPy::setParameters isn't a setter. 

It fails to clear existing content before adding/modifying what is already there. 

A setter need to clear the dict first. 

Due to limited use of Path.Commands beyond simple calls to the constructor this seems not to have created any active bugs and has likely been wrong for ten year since it was written. 

### Full version info

```shell
There is a remaining problem with Command, Command.Parameters() :
 
    
>>> cmd = Path.Command("G0",{"Z":0.0}) 
>>> cmd  
Command ( G0, {'Z':0} )
>>> cmd.Parameters["x"] = 1
>>> cmd.Parameters["y"] = 2
>>> cmd.Parameters["f"] = 3.0
>>> cmd
Command ( G0, {'Z':0} )

Since the following accesses Parameters as a dictionary, Command.setParameters() is not called. (Note lower case keys)

>>> cmd.Parameters
{'Z': 0.0, 'x': 1, 'y': 2, 'f': 3.0}

Now let's call the setter 

>>> cmd.Parameters = cmd.Parameters
>>> cmd
Command ( G0, {'F':3, 'X':1, 'Y':2, 'Z':0} )

Ironically, setting it to itself accesses the params that print() can't see, passes them to the setter which does upcase() on them and replaces all with the new values. 

I do not think direct addition to Command.Parameters is done anywhere due to simple, limited use of Path.Command in FC but it caught me out when trying to add a new gcode param after the initial call to the constructor FreeCAD (exlcusively does.)

If this was a native python object there would not be python object getting out of sync with the underlying CXX data. 

A caveat emptor for any future use of this data structure.
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

J-Dunn, 2025-04-28:
```shell
There is a remaining problem with Command, Command.Parameters() :
 
    
```
>>> cmd = Path.Command("G0",{"Z":0.0}) 
>>> cmd  
Command ( G0, {'Z':0} )
>>> cmd.Parameters["x"] = 1
>>> cmd.Parameters["y"] = 2
>>> cmd.Parameters["f"] = 3.0
>>> cmd
Command ( G0, {'Z':0} )

```
Since the following accesses Parameters as a dictionary, Command.setParameters() is not called. (Note lower case keys)

```
>>> cmd.Parameters
{'Z': 0.0, 'x': 1, 'y': 2, 'f': 3.0}

```

Now let's call the setter 

```
>>> cmd.Parameters = cmd.Parameters
>>> cmd
Command ( G0, {'F':3, 'X':1, 'Y':2, 'Z':0} )
```

Ironically, setting it to itself accesses the params that print() can't see, passes them to the setter which does upcase() on them and replaces all with the new values. 

I do not think direct addition to Command.Parameters is done anywhere due to simple, limited use of Path.Command in FC but it caught me out when trying to add a new gcode param after the initial call to the constructor FreeCAD (exlcusively does.)

If this was a native python object there would not be python object getting out of sync with the underlying CXX data. 

A caveat emptor for any future use of this data structure.
```

J-Dunn, 2025-04-28:
While we are about python-like behaviour, shouldn't print(commandlist[-1]) produce something which looks like a python dict?
https://forum.freecad.org/viewtopic.php?t=96554#p823740

Current output:

print(commandlines[i])
Command G0 [ F:0 X:0 Y:0 Z:11 ]

IMO the Parameters part of that output should be a curly bracketed, comma separate list of key:value pairs. Currently it is neither list nor dict and no commas. Neither is it valid Gcode. What is this supposed to be ?

The following, however, prints correctly since it references a python dict.
commandlines[i].Parameters

May I should add something like this to the current PR :

using namespace Path;

// returns a string which represents the object e.g. when printed in python
std::string CommandPy::representation() const
{
    std::stringstream str;
    str.precision(5);
    str << "Command ( ";
    str << getCommandPtr()->Name;
    str << ", {";
    for (std::map<std::string, double>::iterator i = getCommandPtr()->Parameters.begin();
         i != getCommandPtr()->Parameters.end(); ++i) {
        std::string k = i->first;
        double v = i->second;
        str << "\'" << k << "\':" << v;
        if ( i != std::prev(getCommandPtr()->Parameters.end())  ) str << ", ";
    }
    str << "} )";
    return str.str();
}

>>> params = {"X":0.0 , "Y":2.1}
>>> cmd = Path.Command("G2",params)
>>> params
{'X': 0.0, 'Y': 2.1}
>>> cmd
Command ( G2, {'X':0, 'Y':2.1} )
>>> type(cmd)
<class 'Path.Command'>

It is unclear why this output includes the type ( which is not the the type ) as well.

Isn't the point of PYCXX to produce animals which behave like pythons?

J-Dunn, 2025-04-28:
@wwmayer   @yorikvanhavre

FEA-eng, 2026-05-24:
It's pretty much a code quality issue, so it should still be relevant.

sliptonic, 2026-09-22:
Duplicate.  #21070 has gotten more attention.

closed 2026-09-22 by sliptonic