FreeCAD/FreeCAD #21040: FREECAD CRASHING ON WINDOWS 11 / PART BECOMES HEAVY WHEN REFINING THE PART DESIGN BENCH

NEUFUS, 2025-04-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

What is happening:

I have a problem with this attached part in the refinement section.

The drawing was made in Part Design:

When I use the Sweep and Fillet tools, the drawing becomes heavy, and I can no longer work on the part. When I close the file and try to open it, it takes about 20 minutes or more to open the part. The part is simple. I searched the Forum and Telegram group, but no one was able to help with the problem. I believe the problem must be related to the Windows 11 operating system and graphics card. All files are attached.

[image]

Relatório de informações do Sistema NVIDIA criado em: 04/28/2025 08:50:11
Nome do sistema: DESENVOLVIMENTO

[Monitor]
Versão do DirectX:	12.0 
Processador da GPU:		NVIDIA GeForce RTX 2060
Versão do driver:	576.02
Tipo de driver:		DCH
Nível de recursos do Direct3D:	12_1
Cores CUDA:		1920 
Clock do núcleo:	1680 MHz 
Taxa de dados da memória:	14.00 Gbps
Interface da memória:	192bits 
Largura de banda da memória:	336.05 GB/s
Memória total de gráficos disponível:	21396 MB
Memória de vídeo dedicada:	6144 MB GDDR6
Memória de vídeo dedicada:	0 MB
Memória compartilhada do sistema:	15252 MB
Versão BIOS do vídeo:	90.06.2E.40.F8
IRQ:			Not used
Barramento:		Barramento desconhecido
ID do dispositivo:	10DE 1F08 37591028
Número da peça:		G160 0042

[Componentes]

nvui.dll		8.17.15.7602		NVIDIA User Experience Driver Component

-----------------------------------------------------------------------------------------------

nvxdplcy.dll		8.17.15.7602		NVIDIA User Experience Driver Component
nvxdbat.dll		8.17.15.7602		NVIDIA User Experience Driver Component
nvxdapix.dll		8.17.15.7602		NVIDIA User Experience Driver Component
NVCPL.DLL		8.17.15.7602		NVIDIA User Experience Driver Component
nvCplUIR.dll		8.1.940.0		NVIDIA Control Panel
nvCplUI.exe		8.1.940.0		NVIDIA Control Panel
nvViTvSR.dll		32.0.15.7602		NVIDIA Video Server
nvViTvS.dll		32.0.15.7602		NVIDIA Video Server
nvDispSR.dll		32.0.15.7602		NVIDIA Display Server
nvDispS.dll		32.0.15.7602		NVIDIA Display Server
nvLicensingS.dll		6.14.15.7602		NVIDIA Licensing Server
nvWSSR.dll		32.0.15.7602		NVIDIA Workstation Server
nvWSS.dll		32.0.15.7602		NVIDIA Workstation Server
nvDevToolSR.dll		32.0.15.7602		NVIDIA Licensing Server
nvDevToolS.dll		32.0.15.7602		NVIDIA 3D Settings Server
PhysX		09.23.1019		NVIDIA PhysX
NVCUDA64.DLL		32.0.15.7602		NVIDIA CUDA 12.9.40 driver
nvGameSR.dll		32.0.15.7602		NVIDIA 3D Settings Server
nvGameS.dll		32.0.15.7602		NVIDIA 3D Settings Server
---------------------------------------------------------------------------------------
[image]
[image]

[image]
[image]
[image]
https://github.com/user-attachments/assets/39447ead-43de-48b4-98f8-f5cef5a7fc7c

### Full version info

```shell
FREECAD 1.0

PROBLEM IN REFINING THE PART
BENCH: PART DESIGN
TOOLS: SWEEP AND FILLET
OPEN: ABOUT 20 MINUTES AND THE PART BECOMES HEAVY AND STUCK
```

### Subproject(s) affected?

Assembly

### Anything else?

https://github.com/user-attachments/assets/907c9547-284c-49fb-a486-9590b684c328

[PEÇAS FREECAD.zip](https://github.com/user-attachments/files/19939633/PECAS.FREECAD.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

NEUFUS, 2025-04-28:
[image]

Roy-043, 2025-04-29:
https://forum.freecad.org/viewtopic.php?p=824270#p824270:

> the tessellation setting is way to fine. set it to a normal (> 0.1% and >15deg) or to the default setting, then it will work.

NEUFUS, 2025-04-29:
> https://forum.freecad.org/viewtopic.php?p=824270#p824270:
> 
> > a configuração de mosaico é muito boa. Configure-o para um normal (> 0,1% e >15 graus) ou para a configuração padrão, então ele funcionará.

The problem is that the drawing is heavy, I can't work on the piece. When I use the Sweep command and start refining it starts to crash. The piece is simple, this shouldn't happen.

closed 2025-04-30 by maxwxyz