FreeCAD/FreeCAD #21044: Sketcher: OVP - tab does not switch input box

maxwxyz, 2025-04-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

On testing https://github.com/FreeCAD/FreeCAD/pull/20665 there seems to be issues in the Sketcher OVPs in current main:
1. Tab is only registered when the mouse cursor is absolutely still. As soon as there is minimal movement, tab is ignored
2. If you press tab, enter a value in the second OVP and press tab again, the original input box is not in focus. The user is not able to see what is in focus. When entering numbers, they are added after the already set dimension and change it. Example: Create a line, press tab to set the angle first. Press tab again, there is no option to input the length of the line.

Demo:

https://github.com/user-attachments/assets/e77e1dbe-54b7-4f0d-a41c-6a5f06360cee

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41365 (Git)
Build date: 2025/04/22 11:42:50
Build type: Release
Branch: main
Hash: 84b57d67b50ac64be2246418fb4b49406904f3a9
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/133.772/2
Installed mods: 
  * Cables 0.1.4
  * CfdOF 1.30.1
  * CurvedShapes 1.0.13
  * Curves 0.6.61
  * dodo 1.0.1
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * Gridfinity 0.11.7
  * Manipulator 1.5.9
  * OpticsWorkbench 1.2.0
  * parts_library
  * Rocket 4.1.1
  * sheetmetal 0.7.22
  * Silk 0.1.5
  * Telemetry 1.0.0
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-04-28:
@PaddleStroke FYI

luzpaz, 2025-05-14:
Any diagnosis ?

tetektoza, 2025-06-08:
I've checked it, so second point is probably resolved by my latest change: https://github.com/FreeCAD/FreeCAD/pull/20925
You can check later @maxwxyz in the weekly build just to be sure.

First issue - so based on short analysis, well not maybe short but still I think it's solveable, it looks like our `EventFilter` does not process keyboard events if we flood it with mouse events. Basically you have to stop moving mouse to get this event. I see that
`bool ShortcutManager::eventFilter(QObject *o, QEvent *ev)` is receiving the keyboard events, maybe it's possible to pass from ShortcutManager to `void ViewProvider::eventCallback(void * ud, SoEventCallback * node)` somehow? Need to check, but leaving these observations for future.

maxwxyz, 2025-06-08:
confirm first is fixed, updated the issue

FEA-eng, 2025-12-16:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20251209 (Git shallow)
Build date: 2025/12/09 13:53:17
Build type: Release
Branch: grafted,grafted
Hash: 5ad6a1ba58c665c23f1d217c23e74379d8c1a5a4
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
```