FreeCAD/FreeCAD #21048: BIM: TypeError: unsupported operand type(s) for ^: 'float' and 'float' on creating a wave profile

rhabacker, 2025-04-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When creating a wave profile in FreeCAD 1.0.0, I get a red exclamation mark on the plate object. An error message appears in the output window

TypeError: unsupported operand type(s) for ^: 'float' and 'float'

### Full version info

```shell
OS: openSUSE Leap 15.6 (KDE/plasma5/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.6.15, Qt 5.15.12, Coin 4.0.3, Vtk 9.4.1, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * FreeCAD_FastenersWB-last 0.5.37
  * Assembly4 0.50.15
  * NikraDAP 2.0.0
  * CAM
```

### Subproject(s) affected?

None

### Anything else?

The callstack displayed in the output window shows https://github.com/FreeCAD/FreeCAD/blob/8d6547d519ffb331cfba054b4db2750b3e71339b/src/Mod/BIM/ArchPanel.py#L315 as the code location where the exception occurs.

https://stackoverflow.com/questions/34258537/python-typeerror-unsupported-operand-types-for-float-and-int indicates that Python's exponentiation operator is **, not ^.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ElementW, 2025-04-28:
This is indeed wrong and should be using `**` instead. This was introduced in 710692718c605a66a6024975b3a204106fb9d6f8 and most likely was never tested.

Can reproduce on 1.1.0, latest 1.1.0dev AppImage, and b14d3a224b89c1ffbbce86ef6e8314518e24ef10.

furgo16, 2025-04-29:
@rhabacker @ElementW thanks for reporting this and the analysis. It seems to need a one-liner fix. Would any of you be willing to submit a PR with the fix? Thanks!

PR #21057 (2025-04-29): BIM: Fix Python syntax error

Due to the use of an incorrect exponentiation operator the following exception was triggered when changing attributes of a wave profile:

TypeError: Unsupported operand type(s) for ^: 'float' and 'float'

## Issues

fixes #21048

## Before and After Images

Before: Changing the attribute “wavelength” has no effect, but leads to the exception mentioned.
 [image]
After: The change to the “Wavelength” attribute is displayed accordingly.
[image]
commit 21a52613 (2025-04-29): BIM: Fix Python syntax error

Due to the use of an incorrect exponentiation operator the following
exception was triggered when changing attributes of a wave profile:

TypeError: Unsupported operand type(s) for ^: 'float' and 'float'

fixes #21048
closed 2025-04-29 by chennes