FreeCAD/FreeCAD #21052: Weekly 2025-04-28 does not start on Windows

maxwxyz, 2025-04-29:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

There is an issue with the latest weekly build in the main repository.
The Windows builds do not open with the following error:
[image]

### Full version info

```shell
https://github.com/FreeCAD/FreeCAD/releases/tag/weekly-2025.04.28
Windows
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-04-29:
@oursland FYI

oursland, 2025-04-29:
Looks like [Qt6](https://stackoverflow.com/questions/77010020/how-to-find-qtblurimage-attribute-on-pyqt6-dll-which-is-required-for-neon-effe) has altered the name of the attribute slightly.  Now we must figure out why it was compiled against on definition, but loading against another and failing.

closed 2025-05-01 by hyarion

celdorwow, 2025-05-03:
I was trying to test [Weekly Build weekly-2025.05.02](https://github.com/FreeCAD/FreeCAD/releases/tag/weekly-2025.05.02) Pre-release and don't know if this build was going to fixed the issue. However, in my case I am still having this error:

[image]