FreeCAD/FreeCAD #21064: About Customized Tool Bars

fretteDesBois, 2025-04-29:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Trying to customize a tool bar - erased after a disaster with the Report View, and so deleting  my 2 config and file.sys files - just a remark
The customize a Tool Bar, the left tab says "All", without anything else.
It's pretty clear.
The right tab shows an icon "Global", with an FC image.
I tried this possibility, without an obvious result, surely because I didn't follow the good method to get the good result.
So I sent a message in a bottle, on the FC forum.

So, with an "All", in the right tab (the list of all WBs in which this toll bar will appear), that would give a more consistent interface, saying the same thing on 2 neighbour tabs, when the tabs have to say the same thing : that would make things easier for the user, who don't create customized tool bars 24/7, or I am wrong ?

### Full version info

```shell
Trying to customize a tool bar - erased after a disaster with the Report View, and so deleting  my 2 config and file.sys files - just a remark
The customize a Tool Bar, the left tab says "All", without anything else.
It's pretty clear.
The right tab shows an icon "Global", with an FC image.
I tried this possibility, without an obvious result, surely because I didn't follow the good method to get the good result.
So I sent a message in a bottle, on the FC forum.

So, with an "All", in the right tab (the list of all WBs in which this toll bar will appear), that would give a more consistent interface, saying the same thing on 2 neighbour tabs, when the tabs have to say the same thing : that would make things easier for the user, who don't create customized tool bars 24/7, or I am wrong ?
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-04:
@fretteDesBois in the 'Full version info' section above we intend for that to have the full [About](https://wiki.freecad.org/About) info.

luzpaz, 2026-01-07:
Closing, OP hasn't responded even once.