FreeCAD/FreeCAD #21069: Techdraw: PR19753 regression - files containing Axometric dimensions are not saved correctly causing the file to error on opening and work to be lost

Syres916, 2025-04-30:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Specifically Git Bisect found https://github.com/FreeCAD/FreeCAD/commit/2bbf6c7f6e5146c955f3d7110cb035a42e8ab30a to cause the issue as described in detail in https://forum.freecad.org/viewtopic.php?p=824451#p824256

The attached file was saved with a build prior to the above commit so opens error free but if saved as a different filename using the above commit there is no error on save but when subsequently opened the page is blank and an error in the Report View:

`Invalid Document.xml: End of document reached`

The tree is also corrupted:

[image]

[Support_domad_Syres_with_axonometric_dim.FCStd.zip](https://github.com/user-attachments/files/19970992/Support_domad_Syres_with_axonometric_dim.FCStd.zip)

### Full version info

```shell
Hash: 2bbf6c7f6e5146c955f3d7110cb035a42e8ab30a
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-04-30:
@benj5378 @WandererFan FYI

Syres916, 2025-05-01:
I'm making slow progress, the failing **Document.xml** has two entries for `Tag value=` per `Cosmetic Vertex` when it should have one `VertexTag value=` and one `Tag value=` IMHO. See the extract form the working file:

```
<Property name="CosmeticVertexes" type="TechDraw::PropertyCosmeticVertexList" status="134217728">
<CosmeticVertexList count="2">
<CosmeticVertex type="TechDraw::CosmeticVertex">
<Point X="11.3137084989847629" Y="34.9873428622083154" Z="0.0000000000000000"/>
<Extract value="0"/>
<HLRVisible value="0"/>
<Ref3D value="-1"/>
<IsCenter value="0"/>
<Cosmetic value="1"/>
<CosmeticLink value="-1"/>
<CosmeticTag value=""/>
<VertexTag value="a11172d1-d211-4b1d-993e-68d83a28c40e"/>
<PermaPoint X="11.3137084989847629" Y="34.9873428622083154" Z="0.0000000000000000"/>
<LinkGeom value="49"/>
<Color value="#000000"/>
<Size value="3.5000000000000000"/>
<Style value="1"/>
<Visible value="1"/>
<Tag value="a11172d1-d211-4b1d-993e-68d83a28c40e"/>
</CosmeticVertex>
```

and the same extracted portion of the failing file:

```
<Property name="CosmeticVertexes" type="TechDraw::PropertyCosmeticVertexList" status="134217728">
<CosmeticVertexList count="2">
<CosmeticVertex type="TechDraw::CosmeticVertex">
<Point X="11.3137084989847629" Y="34.9873428622083154" Z="0.0000000000000000"/>
<Extract value="0"/>
<HLRVisible value="0"/>
<Ref3D value="-1"/>
<IsCenter value="0"/>
<Cosmetic value="1"/>
<CosmeticLink value="-1"/>
<CosmeticTag value=""/>
<Tag value="a11172d1-d211-4b1d-993e-68d83a28c40e"/>
<PermaPoint X="11.3137084989847629" Y="34.9873428622083154" Z="0.0000000000000000"/>
<LinkGeom value="51"/>
<Color value="#000000"/>
<Size value="3.5000000000000000"/>
<Style value="1"/>
<Visible value="1"/>
<Tag value="a11172d1-d211-4b1d-993e-68d83a28c40e"/>
</CosmeticVertex>
```

@benj5378 am I on the right track here?

benj5378, 2025-05-01:
@Syres916 I don't think you're on the right track. The commit referenced was to centralize the tag functionality, indirectly renaming the tag element in the XML.

I think the issue is the commit assuming element to be named `tag` when it is named `VertexTag` in old files. There should have been a migration script.

WandererFan, 2025-05-02:
What about handling variations in attribute names like this:

```
void Tag::Restore(Base::XMLReader& reader)
    if (reader.readNextElement()) {
        if(strcmp(reader.localName(),"Tag") == 0 ||
           strcmp(reader.localName(), "VertexTag") == 0) ||
           strcmp(reader.localName(), "ADifferentTag") == 0)  {
            std::string temp = reader.getAttribute("value");
            ...
        }
    }
```

WandererFan, 2025-05-15:
Coder notes: 
- fails to read valid Document.xml
    - in Vertex::Restore(), the call to Tag::Restore(reader, "VertexTag") will report reaching end of
      document (and does not load document) if the CosmeticVertex section of Document.xml does not contain a VertexTag element.
    - in current main branch, the only occurrences of the string "VertexTag" are in Vertex::Save and Vertex::Restore.

- breaks backwards compatibility
    - a file created with v0.21.2 shows both CosmeticTag and VertexTag elements, with VertexTag set to a
      uuid and CosmeticTag empty (but cosmetic flag is set).
    - a file created with recent main shows an empty CosmeticTag, 2 identical Tag and 0 VertexTag
      elements.  The cosmetic flag is set.
    
    - in v0.21 and 1.0, Vertex::Save() outputs the cosmeticTag member as CosmeticTag and getTagAsString() as
      VertexTag. Vertex::Restore() sets cosmeticTag from CosmeticTag element and tag from VertexTag.
      
    - Vertex::getCosmeticTag() returns the value of the cosmeticTag member, not Tag.
    - there is code in CosmeticExtension that relies on getCosmeticTag() to match CVs. Not sure how that 
    - is working with an empty CosmeticTag. Should getCosmeticTag() return Tag??

    - there are documents out there with VertexTag and also some without VertexTag, so a fix should 
      handle both.

senjabl, 2025-06-11:
Seems ok now. Few files I saved in broken state before this PR stayed corrupted but fortunately I have working backups