FreeCAD/FreeCAD #21070: CAM Command::setParameters is not a setter.

J-Dunn, 2025-04-30:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Command::setParameters takes one argument, a dict. 

Instead of replacing the existing dict, as a setter should do , it simply adds/replaces existing content. 

This means that Command has not setter and that it is impossible to delete any of the existing data

### Full version info

```shell
current master. 
Appears to be the case since its inception 10y ago. No recent changes.
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-07:
Can anybody weigh-in on this ?

knipknap, 2025-05-19:
Interestingly, this method appears to be completely unused, so I think it should be safe to apply a fix? Or just remove the whole method...

```bash
freecad-source$ grep -R "setParameters" src/Mod/CAM/
src/Mod/CAM/App/CommandPyImp.cpp:// setParameters() or other objects invalidate its content, eg setPlacement()
src/Mod/CAM/App/CommandPyImp.cpp:void CommandPy::setParameters(Py::Dict arg)
```

J-Dunn, 2025-05-22:
eg theadmilling.py 

```
                for cmd in commands:
                    p = cmd.Parameters
                    if cmd.Name in ["G0"]:
                        p.update({"F": self.vertRapid})
                    if cmd.Name in ["G1", "G2", "G3"]:
                        p.update({"F": self.horizFeed})
                    cmd.Parameters = p

```

here a copy is taken, altered, then copied back. In this case they get away with adding instead of replacing Parameters since the other remain unchanged. Phew!

J-Dunn, 2025-05-23:
To turn this into a correctly functionning setter instead of an adder, Parameters needs to be cleared before the supplied arg is copied in. 

```
void CommandPy::setParameters(Py::Dict arg)
{
    PyObject* dict_copy = PyDict_Copy(arg.ptr());
    PyObject *key, *value;
    Py_ssize_t pos = 0;
    getCommandPtr()->Parameters.clear();

```

tarman3, 2026-02-28:
C++ for me is like the green screen in film Matrix
Sorry

I do not see any women in red dress here ))
[image]

davidgilkaufman, 2026-03-09:
> I'm happy to review/approve a 1-line cpp diff to fix this problem

I suppose barring input from someone who knows (or learns) what dxf_post is supposed to do, I'd also merge a 1 line diff plus a change to dxf_post to preserve current behavior and flag the code with an explanatory comment

FEA-eng, 2026-09-16:
I assume that it's still relevant.

luzpaz, 2026-09-16:
@J-Dunn are you planning to make a PR ?

sliptonic, 2026-09-22:
> [@J-Dunn](https://github.com/J-Dunn) are you planning to make a PR ?

After six months, obviously not.  I've submitted a pr and requested a review from @davidgilkaufman

iromero91, 2026-09-24:
@sliptonic tying in context from 4 years ago on this issue, see #6310

closed 2026-09-29 by Connor9220