FreeCAD/FreeCAD #21071: techdraw outputs the dimensions incorrectly

Tabintaban, 2025-04-30:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In the FreeCAD 1.0 version, when printing a drawing, there is no saving of 1:1 dimensions.
From the TechDraw workbench and subsequent saving to SVG, the dimensions of the drawing are reduced.
The distance of 10.5 cm actually turns out to be 9 cm.

### Full version info

```shell
In the FreeCAD 1.0 version, when printing a drawing, there is no saving of 1:1 dimensions.
From the TechDraw workbench and subsequent saving to SVG, the dimensions of the drawing are reduced.
The distance of 10.5 cm actually turns out to be 9 cm.
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-04-30:
You need to attach an example file. Just add .zip to the filename before uploading.

Please also add your full version info:
https://wiki.freecad.org/Std_About > Copy to clipboard > Paste result in text field.

WandererFan, 2025-04-30:
Svg export scaling looks correct here:
[image]

Pendelf, 2025-05-02:
I have it exporting fine with the same dimensions. Attach the file

https://github.com/user-attachments/assets/56a06f96-b80b-46e3-ae10-dd620827bda9

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41365 (Git)
Build date: 2025/04/22 11:42:50
Build type: Release
Branch: main
Hash: 84b57d67b50ac64be2246418fb4b49406904f3a9
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * AirPlaneDesign 0.4.1
  * btl 0.9.9
  * CurvedShapes 1.0.13
  * Curves 0.6.61
  * fasteners 0.5.38
  * PieMenu 1.11.3
  * Road 2025.2.22
  * sheetmetal 0.7.22
  * Telemetry 1.0.0

luzpaz, 2025-05-03:
@Tabintaban we also need your full [about](https://wiki.freecad.org/About) info

FEA-eng, 2025-10-08:
@Tabintaban Can you confirm if it's still relevant in the newest weekly build available [here](https://github.com/FreeCAD/FreeCAD/releases)?