FreeCAD/FreeCAD #21072: fermeture de freeecad

lion56270, 2025-04-30:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Freecad se ferme tout seul lors de l'utilisation sur ubuntu 24.04

### Full version info

```shell
OS: Ubuntu Core 22 (MATE/mate/xcb)
Architecture: x86_64
Version: 1.0.0.38641 +468 (Git) Snap 1248
Build type: Release
Branch: tag: 1.0.0
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * woodworking 0.21.2.33771.1.0.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ElementW, 2025-04-30:
Please write in English so everybody here can understand you.

In addition, please provide additional information, as your bug reports conveys *zero* clue about what is happening. FreeCAD is closing yes, but what triggers this close? Is it random, is it a particular action, is it a particular file?

maxwxyz, 2025-04-30:
Please check out how to [report issues](https://github.com/FreeCAD/FreeCAD?tab=readme-ov-file#reporting-issues) which could help us diagnose and fix it.

- Test if the issue is still present in the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
- add full [About](https://wiki.freecad.org/About) info to the issue (`Help > About FreeCAD > Copy to clipboard`).
- share an example file with the issue present. You can upload a zipped file to the original issue.
- add steps to reproduce the issue. Ideally, provide a macro of the issue: `Macro > Macro recording...` and repeat all steps until the issue occurs.

Feel free to reopen the issue after adding more details.

closed 2025-04-30 by maxwxyz