FreeCAD/FreeCAD #21077: Can't select fonts from hidden directories

macdroid53, 2025-04-30:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

On Linux it is possible to have various things in "hidden" directories, typically in the home directory there are both .config and .local.

When creating a Shapestring the font selection dialog seems to have no way to expose (typically ctrl-h) these dot directories.

To reproduce, attempt to create a Shapestring, in the dialog attempt to change the font that resides in .local or .config directory.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41485 (Git)
Build date: 2025/04/30 03:56:09
Build type: Release
Branch: main
Hash: 14d276b120235866b67182f4476347849e167ad6
Python 3.13.2, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, boost 1_87, Eigen3 3.4.0, PySide 
shiboken , SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/93.6146/1
Installed mods: 
  * Silk 0.1.5
  * Defeaturing 1.2.4
  * Curves 0.6.61
  * CurvedShapes 1.0.13
  * DynamicData 2.74.0
  * lattice2 1.0.0
  * sheetmetal 0.7.22
  * Manipulator 1.5.9
  * kicadStepUpMod 11.6.8
  * fasteners 0.5.38
  * OpenTheme 2024.9.1
  * ProDarkThemePreferencePack 1.0.0
  * Channels-main 0.1.0.dev2
  * freecad.gears 1.3.0
  * ExtremeProDark 2.9.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-04-30:
Is this with the native file dialog or with the non-native file dialog?
https://wiki.freecad.org/Draft_ShapeString#Preferences

Syres916, 2025-04-30:
I thought when the user activates the ShapeString command `DontUseNativeDialog` is temporarily set to `true` by default.

Roy-043, 2025-04-30:
If I right-click in the file panel I can check "Show hidden files".

[image]

Roy-043, 2025-04-30:
> It is Ctrl-h that does not work. If you right click in the dialog it offers a "Show hidden" checkbox.

I just found that out myself. I think we can say that the original issue has been solved then.

maxwxyz, 2025-05-01:
So the issue is that a custom or system shortcut does not work in a system file dialog?

macdroid53, 2025-05-01:
> So the issue is that a custom or system shortcut does not work in a system file dialog?

Yes

bstott, 2025-05-21:
I have this issue.  A ShapString is created via Draft WB. The selection of Font works within the setup dialogue.  Close the ShapeString dialogue. Select the ShapString in the Tree - go to the Data Tab - Select the Font File "..." The Windows Font directory opens with NOTHING VISIBLE. So, a person can't change the font via the Data Tab.

Roy-043, 2025-05-22:
@bstott Please read:
https://wiki.freecad.org/Draft_ShapeString#Font_file_selection_on_Windows

bstott, 2025-05-23:
It doesn't seem to be an answer. But, an excuse.  From a programming, engineering, technical, end user perspective -  IF, you can get a list of available fonts displayed when setting up the Feature and while creating a Scripture - Then, transition into adjustment of the attributes of the created feature object - The Data Tab should avail ALL the same FONTS, which the user had access to initially. IF, the fints were viewed and selected initially then, they should be available to view and change later.  The user shouldn't have to make adjustments, configuration changes, etc. Or, be given excuses. Which apparently didn't exist a moment prior. EH?  Just saying.

Roy-043, 2025-05-23:
Alternatively you can double-click the ShapeString in the Tree view.

> It doesn't seem to be an answer. But, an excuse.

Until we have a way to handle a file property based on the file type, or until Windows removes the mentioned restrictions (unlikely), this is the best we can do.

luzpaz, 2026-09-04:
What's the most accurate title for this ticket in that case ?