FreeCAD/FreeCAD #21086: CAM: Adaptive cuts same depth many times

jffmichi, 2025-04-30:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The adaptive operation cuts the same depth many times. E.g. the last depth -20mm is cut 120 times.

FreeCAD project file:
[adaptive_multiple_depth_issue.zip](https://github.com/user-attachments/files/19985621/adaptive_multiple_depth_issue.zip)

Resulting GCode:
[adaptive_multiple_depth_issue_Job.zip](https://github.com/user-attachments/files/19985636/adaptive_multiple_depth_issue_Job.zip)

### Full version info

```shell
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.41450 (Git)
Build date: 2025/04/28 08:04:40
Build type: Release
Branch: makepkg
Hash: ed85cd3a5d50fa1b15678e574ba2d943581206c9
Python 3.13.3, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, boost 1_87, Eigen3 3.4.0, PySide 
shiboken , SMESH 7.7.1.0, xerces-c 3.3.0, OCC 8.0.0.rc1-7aa85582ad
Locale: English/United States (en_US) [ OS: German/Germany (de_DE) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/breeze
Logical DPI/Physical DPI/Pixel Ratio: 96/93.1131/2
Installed mods: 
  * kicadStepUpMod 11.5.7
  * Gridfinity 0.11.4
  * dodo 1.0.1
  * sheetmetal 0.7.22
  * fcgear 1.0.0
  * fasteners 0.5.37
  * Assembly4 0.50.18
  * Curves 0.6.60
  * lattice2 1.0.0
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

jffmichi, 2025-05-01:
@dbtayl since you recently made changes to the Adaptive operation, could you maybe have a look? Thank you in advance :)

dbtayl, 2025-05-01:
Seems to work fine on 41354, broken on or before apparently 41450. I won't have time to look deeper for at least several hours, but a quick glance at the commits in the CAM module don't show anything obvious. Also not obviously related to the adaptive changes I made, at least not directly.

dbtayl, 2025-05-02:
How extreme the problem is seems to vary with the stepdown- my previous comment isn't actually accurate, it happens even on older versions.

I think this one is related to the adaptive changes I made. Not enough time to fully isolate/fix it tonight, but looks like the code that nests different regions to machine is duplicating the same region when each should only appear once.

PR #21177 (2025-05-12): CAM: Fix adaptive repeating cuts excessively at various depths

Correctly handle regions contained entirely within the stock to avoid recutting them multiple times. Root cause was regions fully contained within the stock not being correctly computed as "the same", resulting in them being re-computed, along with the region at every lower depth, resulting in a factorial(ish) number of duplicate cuts.

## Issues

Fixes #21086.
## Before and After Images

N/A