FreeCAD/FreeCAD #21091: On copy, don't preselect App::Link sources

NomAnor, 2025-05-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When copying (duplicating) `App::Link` objects, the copy dialog automatically select the link sources as well. In my opinion, this is not what is typically expected. Especially if I copy an ``App::Part`` Assembly with a bunch of links in it.

Only the link should be copied, and the result should point to the same part as the original link.

[image]

### Full version info

```shell
OS: Arch Linux (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.41480 (Git)
Build date: 2025/04/29 12:20:12
Build type: Unknown
Branch: main
Hash: 1bf21f85bd51f9347aa3ba99994fe1d4f3cb4e92
Python 3.13.3, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, boost 1_87, Eigen3 3.4.0, PySide 
shiboken , SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/141.585/1
Installed mods: 
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * lattice2 1.0.0
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

entuland, 2026-08-27:
A similar scenario is a body which depends on something else outside its subtree (spreadsheet, varset, freestanding sketches or other shapes, depended on via binders), the user may want to copy the entire body subtree and omit all the dependencies outside the subtree.

So this could be generalized to giving a simple way to copy an item with all its subtree, without having to cherry pick the selection (be it in the tree before copying or inside the copy dialog)

kinghat, 2026-08-27:
if i want to duplicate/copy an object in the tree i would expect the default to be that object and everything under it, recursively, and nothing above it. anything outside of that would be a more "advanced" scenario.