FreeCAD/FreeCAD #21094: Weekly 2025-04-28 freezes on macOS

maxwxyz, 2025-05-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I can start FreeCAD normally but as soon as a dialog pops up or I open a menu or right click, it freezes indefinitely on macOS.
I cannot access the about info because as soon as I click on the menu item it hangs. Also in safe mode.

### Full version info

```shell
OS: macOS 13.7.3
Architecture: x86_64
Weekly 2025-04-28
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-01:
@oursland FYI, might be related to the Windows issue but it seems to run fine on arm.

oursland, 2025-05-01:
I have the same issue, and I'm on ARM64 (Apple M2 Max).  The weekly builds appear to work for some, but not for others.  I suspect a configuration issue, but the cause is unknown.

depthoffocus, 2025-05-04:
Is this connected to #21014 ?

I am seeing these hangs with the 2025-05-02 weekly, but mostly when the hovertext/tooltip is on the screen.

I cannot load a file from the recent files menu without a hang, for example, because a tooltip will appear. If I get to the File > Open menu and choose a file and open it, the hang won't appear until the first hovertext appears.

Sometimes it is impossible to get to the menu to do anything before triggering this.

I suspect the hang sometimes happens before the hovertext is actually rendered.

It's not quite the same as the weird long delays that happen when copying the version details to the clipboard, which I have a feeling might be to do with the unsigned app negotiating OS permissions, because the delay in *that* case lasts exactly as long as the (quite long) delay which is triggered by the unsigned weekly releases when you try to access Documents/Downloads etc.

oursland, 2025-05-04:
The weekly builds on `FreeCAD/FreeCAD` are now signed and notarized, so we should no longer have any issues due to Apple Gatekeeper.  You will still get the notice that the application was downloaded from the internet, but that is due to the quarantine attribute being set and the user will be able to click the continue button.

eendeego, 2025-05-05:
I keep getting the dialog for (non signed ?) apps every time I try to open weekly builds.

> “FreeCAD-b41488.app” is damaged and can’t be opened. You should move it to the Trash.

JohnOCFII, 2025-05-05:
> I keep getting the dialog for (non signed ?) apps every time I try to open weekly builds.
> 
> > “FreeCAD-b41488.app” is damaged and can’t be opened. You should move it to the Trash.

The FreeCAD build process is migrating from FreeCAD-bundle (the legacy builds) to releases that will be in the main FreeCAD repository.  The **new** releases in the main repository **ARE** signed.  But the website still refers to the older version, as there are still some teething issues in the new version.  So, in the meantime, the versions in the `FreeCAD-bundle` repository will still require executing `xattr` to clear the quarantine bit.

``` xattr -cr /Applications/FreeCAD*```

eendeego, 2025-05-05:
Thank you! That works for me! Should that command be on the weekly downloads text ?

JohnOCFII, 2025-05-05:
> Thank you! That works for me! Should that command be on the weekly downloads text ?

In hindsight, we should have done that 4 months ago... and now we are very close (a couple weeks?) to the new weekly build process where the downloads will all be properly signed, so you won't need to go through this process.

hsvfan-jan, 2025-05-06:
I was trying to downgrade versions to get rid of this bug and even with `Weekly Build weekly-2025.04.21`, one version earlier than the one mentioned in the summary, I get the freeze.

I found an older version in the bin that I downloaded from FreeCAD's website iirc and it doesn't have the same bug, even though the date seems to be the same.
```
OS: macOS 15.4.1
Architecture: arm64
Version: 1.1.0dev.41354 (Git)
Build date: 2025/04/21 18:01:03
Build type: Release
Branch: main
Hash: fd390acd53dae91898145a22d74050d0d648a45b
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/Germany (en_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
```

I don't remember what open source project it was, but there it was recommended not to use the GitHub releases and instead the ones from the website. So perhaps also related to how it's built?

oursland, 2025-05-06:
@hsvfan-jan I am working on addressing the freezing issues on macOS.  Today we merged three PRs, which will hopefully resolve this issue:

* https://github.com/FreeCAD/FreeCAD/pull/21164
* https://github.com/FreeCAD/FreeCAD/pull/21163
* https://github.com/FreeCAD/FreeCAD/pull/21166

hsvfan-jan, 2025-05-13:
@oursland Seems to have worked!

No crashes on
```
OS: macOS 15.5
Architecture: arm64
Version: 1.1.0dev.41671 (Git)
Build date: 2025/05/12 16:21:14
Build type: Release
Branch: main
Hash: a997c1ef16a0a4657e0edf61fb15d01058832227
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/Germany (en_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
```

oursland, 2025-05-13:
I, too, confirm that FreeCAD no longer freezes upon startup.  I think we can close this issue.  @maxwxyz can you test and confirm?

maxwxyz, 2025-05-13:
I can confirm too!

closed 2025-05-13 by maxwxyz