FreeCAD/FreeCAD #21095: Weird crash at App::Transaction::addObjectChange

vosk, 2025-05-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I have a macro that fixes a mesh, converts to solid, then deletes mesh object:
```
# -*- coding: utf-8 -*-

import FreeCAD
import Mesh
import Part
import OpenSCADUtils

sel = FreeCADGui.Selection.getSelection()
print(sel)
for x in sel:
	if x.Mesh.CountFacets == 0:
		App.ActiveDocument.removeObject(x.Label)
		continue
	if not x.Mesh.isSolid():
		x.Mesh.fillupHoles(5000)
	if x.Mesh.isSolid():
		s = Part.Shape()
		s.makeShapeFromMesh(x.Mesh.Topology, 0.100000, True)
		s = s.removeSplitter()
		o = App.ActiveDocument.addObject("Part::Feature",x.Label+ "_solid")
		o.Shape = Part.Solid(s)
		App.ActiveDocument.removeObject(x.Name)
App.ActiveDocument.recompute()
```
As part of my workflow I often encountered the following stacktrace SIGSEGV
```
(gdb) bt
#0  0x00007ffff569367a in App::Transaction::addObjectChange(App::TransactionalObject const*, App::Property const*) () at /usr/lib/freecad-daily-python3/lib/libFreeCADApp.so
#1  0x00007ffff62865d4 in Gui::ViewProviderDocumentObject::onBeforeChange(App::Property const*) () at /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so
#2  0x00007ffff571dd87 in App::PropertyEnumeration::setEnums(char const**) () at /usr/lib/freecad-daily-python3/lib/libFreeCADApp.so
#3  0x00007ffff6291db6 in Gui::ViewProviderDocumentObject::attach(App::DocumentObject*) () at /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so
#4  0x00007fff29f0f2bd in MeshGui::ViewProviderFace::attach(App::DocumentObject*) () at /usr/lib/freecad-daily-python3/lib/MeshGui.so
#5  0x00007fff29f0d5f6 in MeshGui::MeshFaceAddition::startEditing(MeshGui::ViewProviderMesh*) () at /usr/lib/freecad-daily-python3/lib/MeshGui.so
#6  0x00007fff29f10f55 in CmdMeshAddFacet::activated(int) () at /usr/lib/freecad-daily-python3/lib/MeshGui.so
#7  0x00007ffff5f4718f in Gui::Command::_invoke(int, bool) () at /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so
#8  0x00007ffff5f4753e in Gui::Command::invoke(int, Gui::Command::TriggerSource) () at /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so
#9  0x00007ffff3312e16 in ??? () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#10 0x00007ffff3f64f94 in QAction::triggered(bool) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#11 0x00007ffff3f67eab in QAction::activate(QAction::ActionEvent) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#12 0x00007ffff406967a in ??? () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#13 0x00007ffff40697b8 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#14 0x00007ffff416d446 in QToolButton::mouseReleaseEvent(QMouseEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#15 0x00007ffff3fb0df8 in QWidget::event(QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16 0x00007ffff3f6bd45 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#17 0x00007ffff3f746b0 in QApplication::notify(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#18 0x00007ffff5f052fe in Gui::GUIApplication::notify(QObject*, QEvent*) () at /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so
#19 0x00007ffff32d8118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#20 0x00007ffff3f72874 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#21 0x00007ffff3fcaa39 in ??? () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#22 0x00007ffff3fcdfbf in ??? () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#23 0x00007ffff3f6bd45 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#24 0x00007ffff5f052fe in Gui::GUIApplication::notify(QObject*, QEvent*) () at /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so
#25 0x00007ffff32d8118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#26 0x00007ffff3745a3b in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#27 0x00007ffff3717bfc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#28 0x00007fffed57bd06 in ??? () at /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#29 0x00007ffff15145b5 in ??? () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#30 0x00007ffff1573717 in ??? () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#31 0x00007ffff1513a53 in g_main_context_iteration () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#32 0x00007ffff3335279 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#33 0x00007ffff32d6a7b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#34 0x00007ffff32df3e8 in QCoreApplication::exec() () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#35 0x00007ffff5e814bf in Gui::Application::runApplication() () at /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so
#36 0x000055555555a3da in ??? ()
#37 0x00007ffff282a1ca in __libc_start_call_main (main=main@entry=0x555555559dd0, argc=argc@entry=1, argv=argv@entry=0x7fffffffdd28) at ../sysdeps/nptl/libc_start_call_main.h:58
#38 0x00007ffff282a28b in __libc_start_main_impl (main=0x555555559dd0, argc=1, argv=0x7fffffffdd28, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffdd18) at ../csu/libc-start.c:360
#39 0x000055555555b4a5 in ??? ()

```

Steps to reproduce:
document with 2 closed meshes (eg 2 cubes, and then mesh -> Create mesh from shape)
select one mesh.
run macro 
select other mesh
click  on Mesh -> Add triangle
crash

Let me know if you need anything more

### Full version info

```shell
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.40971 (Git)
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * MeshRemodel 1.10.37
  * boltsfc 2022.11.5
  * fasteners 0.5.38
  * ThreadProfile 1.93.0
```

### Subproject(s) affected?

Mesh

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

FEA-eng, 2026-01-08:
Still relevant (access violation on Windows)

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260106 (Git shallow)
Build date: 2026/01/06 15:36:19
Build type: Release
Branch: (HEAD detached at 9b64da8)
Hash: 9b64da827a112d88a025be26316e3d023ff491dc
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

luzpaz, 2026-01-08:
@ickby any chance you can check this out ?