FreeCAD/FreeCAD #21096: TechDraw print/Export as PDF strokes some text incorrectly

jethrogb, 2025-05-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When I print a TechDraw (or export it to PDF), the text in the page template block gets rendered weirdly:

[image]

If you open the PDF in e.g. InkScape, you can see that the correct text rendering is there, but also another object on top that appears to be a stroke on top of the original text's path. Other text, such as dimensions or the coordinates on the side of the page don't have this issue.

Export to SVG is not affected.

This happens to both existing projects and new ones.

Sample problem file: [techdraw.pdf](https://github.com/user-attachments/files/20004947/techdraw.pdf)

### Full version info

```shell
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.13.2, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/breeze
Installed mods: 
  * sheetmetal 0.2.49

This appears to be a regression from 1.0.0-3 to 1.0.0-10 (Arch Linux package version). But perhaps it's another system library that got upgraded around the same time.
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-01:
> This appears to be a regression from 1.0.0-3 to 1.0.0-10 (Arch Linux package version). But perhaps it's another system library that got upgraded around the same time.

This is strictly an Arch Linux issue. The FreeCAD Arch packagers need to be notified.

luzpaz, 2025-05-01:
If you can reproduce on 1.1-dev then please let us know and we'll re-open this ticket.

jethrogb, 2025-05-01:
Reported at https://gitlab.archlinux.org/archlinux/packaging/packages/freecad/-/issues/20

alerque, 2025-05-02:
> This is strictly an Arch Linux issue.

I'm one of the Arch Linux maintainers for this package. If this is indeed a packaging issue I'd be more than happy to address it, but it isn't clear to me what that issue would be or even how to start looking for it. We try to keep our packaging as close to upstream sources and defaults as possible.

There is a notable exception in this case where we have backported a fix from from this repository from after 1.0.0 and applied it to 1.0.0 sources to get the build working with current VTK releases. However that backport was done in 1.0.0-3, which the OP here states is before the regression. Everything between PKGREL 4 and 11 has been normal plain rebuilds as other system dependencies get upgraded.

That makes it extremely likely that there is some incompatibility between current versions of some system dependency and the last stable release of this project. Just because your own development HEAD versions don't have this issue doesn't necessarily make it a packaging issue. You may have fixed whatever the problem was since, but the 1.0.0 tag seems to be the last stable release we should be packaging and that version seems to have some incompatibility.

If anybody has any suggestions for what to look into I'm open for trying things, but it isn't transparent to me if/how this is a packaging issue.

luzpaz, 2025-05-08:
Hi @alerque thanks for following up. I've re-opened the issue based on your points.

luzpaz, 2025-05-08:
@adrianinsaval any help would be appreciated https://github.com/FreeCAD/FreeCAD/issues/21096#issuecomment-2846446977

alerque, 2025-05-08:
@jethrogb Can you include a simple source file and font showing this problem, not just the PDF output?

jethrogb, 2025-05-08:
Here's a source file: [print-issue.FCStd.zip](https://github.com/user-attachments/files/20109998/print-issue.FCStd.zip) (you need to remove the .zip extension, that's needed to upload on GitHub).

I'm not sure what you mean by font, I didn't select any font. I'm just using the defaults that come with FreeCAD/the TechDraw template (A4_Landscape_TD).

alerque, 2025-05-08:
I'm able to replicate the font rendering issue, so that's something.

It appears FreeCAD isn't actually managing this template so much as embedding it. As I understand it trying to look into this issue, the template is actually an SVG file and not editable in FreeCAD directly at all. This suggests this probably *is* a font issue, and the FreeCAD GUI vs. the PDF exporter probably aren't using the same SVG rendering code. 

Does anybody know what library is being used in under the hood for rendering the SVG of the template to PDF? Or how I can extract the SVG out of the FreeCAD file to inspect what text properties are being applied?

alerque, 2025-05-09:
@luzpaz In addition to the points above, I hav compiled `1.1.0dev 41623 (GIT)` (from 340cfc6feb) and confirmed that it exhibits the same problem as 1.0.0.

sean9721, 2025-05-17:
I can mostly confirm.  I hope I don't add any confusion.   I noticed this a few months ago and wondered if it was just me experiencing this.

With 1.0, TD pages export to PDF correctly, with no stroke.  _Right click TD page -> Export PDF_.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * DynamicData 2.74.0
  * fasteners 0.5.37
  * OpenTheme 2024.9.1
  * sheetmetal 0.7.21
```

[1.0 File.FCStd.zip](https://github.com/user-attachments/files/20266595/1.0.File.FCStd.zip)  (Remove .zip extension before opening.)

No stroke, as shown below:

[image]

[1.0 TD, Right Click, Export PDF.pdf](https://github.com/user-attachments/files/20266599/1.0.TD.Right.Click.Export.PDF.pdf)

In 1.1-41155, _TD page -> Right click -> Export PDF_ adds a stroke to the text.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41155 (Git) Conda
Build date: 2025/04/06 04:00:07
Build type: Release
Branch: main
Hash: e537278f1296211edb90bf1c938dfaebd2626f0d
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/106.605/1
Installed mods: 
  * DynamicData 2.74.0
  * fasteners 0.5.37
  * OpenTheme 2024.9.1
  * sheetmetal 0.7.21
```

[1.1 File.FCStd.zip](https://github.com/user-attachments/files/20266596/1.1.File.FCStd.zip)  (Remove .zip extension before opening.)

Added stroke shown below:

[image]

[1.1 TD, Right Click, Export PDF.pdf](https://github.com/user-attachments/files/20266598/1.1.TD.Right.Click.Export.PDF.pdf)

With 1.1-41155, _TD page -> Right Click -> Print All Pages -> Print To PDF_ does not add a stroke, but it rasterizes (pixelizes) the exported page.

[image]

[1.1 TD, Right Click, Print All Pages.pdf](https://github.com/user-attachments/files/20266597/1.1.TD.Right.Click.Print.All.Pages.pdf)

alerque, 2025-05-17:
The fact that you are testing on Windows further suggests this is *not* a packaging issue. Also it may not be a Linux issue. Perhaps others can further test on other platforms and/or distros using the above test files to compare results and confirm/deny what works where. That could help narrow down what library/dependency differences are actually relevant.

tobiasfalk, 2025-05-18:
This seams related to https://github.com/FreeCAD/FreeCAD/issues/18881 and my experience is that QT has a buch of problems with PDF Text, especially on Windows.

tobiasfalk, 2025-05-18:
This seams related to https://github.com/FreeCAD/FreeCAD/issues/18881

alerque, 2025-05-19:
Yes, that seems to be the same problem. That makes this issue a duplicate of the earlier one, and the initial diagnosis of this being a distro packaging issue was just a total mis-step. This issue does give some context the other one doesn't have, notably that it seems the same FreeCAD version may have worked properly at some point, so a library version bump is likely the culprit. I've commented to that effect on the other issue. This one can probably be closed as a dup.

WandererFan, 2025-05-19:
<<Does anybody know what library is being used in under the hood for rendering the SVG of the template to PDF? Or how I can extract the SVG out of the FreeCAD file to inspect what text properties are being applied?>>

see PagePrinter::printPdf or PagePrinter::printAllPdf and QGISVGTemplate::load.

The Pdf is created by QPdfWriter.  Svg is rendered by QSvgRenderer.

jethrogb, 2025-05-19:
It could be related, but the output text in the other issue it at least readable.

WandererFan, 2025-08-16:
This is caused by a bug in Qt 6.  It seems to be fixed in 6.8.3 (but not in 6.9.1?).

[image]

OS: Linux Mint 21.3 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/08/12 22:08:35
Build type: Release
Branch: main
Hash: 4195f7fee28e65b289b9390ff7a0deab49e81040
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/Canada (en_CA)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/118.229/1.19792
Installed mods: 
  * fasteners 0.5.21

WandererFan, 2025-08-16:
https://bugreports.qt.io/browse/QTBUG-123817

tobiasfalk, 2025-08-16:
6.9.2 should be just around the corner.
https://wiki.qt.io/Qt_6.9_Release

tobiasfalk, 2025-08-24:
The plan was updated to the 25.8 .

luzpaz, 2025-08-24:
> The plan was updated to the 25.8 .

25.8 ?

WandererFan, 2025-08-24:
6.9.2 doesn't seem to have made it to the Qt Maintenance Tool yet.  I only see options for 6.9.1 and 6.10.0beta.

luzpaz, 2025-08-26:
[image]

Source: https://www.qt.io/offline-installers

tobiasfalk, 2025-09-16:
Has someone compiled FreeCAD with the newest version and Tested if this error still is occurring?

WandererFan, 2025-10-07:
I am running Qt 6.9.3 and it seems to be fixed.

[image]

[Unnamed_Page_DN_AAA_QT6.9.3.pdf](https://github.com/user-attachments/files/22754950/Unnamed_Page_DN_AAA_QT6.9.3.pdf)

FEA-eng, 2026-10-25:
I can't reproduce it, but it's most likely OS-specific.

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 621.01/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```