FreeCAD/FreeCAD #21097: Crashes on boolean cut operation

paranoidi, 2025-05-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

1. Select PlateBody
2. CTRL-Select Threaded Rod from PlateBody
3. In part WB issue boolean cut operation

FreeCAD 1.1dev crashes. I tested this on 1.0 and it works (although opening the same project in 1.0 gives errors).

I tried to use freecad-daily / appimage builds but they are currently in a state where even creating new project fails. The 1.1dev version that I use is few months old since Flatpak does not seem to be regularly updated.

[Holder2.FCStd.zip](https://github.com/user-attachments/files/20005001/Holder2.FCStd.zip)

### Full version info

```shell
04d2bcbf45ab96f3895ac3d0e8b5ae9becbaf5cd
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Pendelf, 2025-05-02:
I confirm, with the thread on it doesn't work, with it off it does

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41365 (Git)
Build date: 2025/04/22 11:42:50
Build type: Release
Branch: main
Hash: 84b57d67b50ac64be2246418fb4b49406904f3a9
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * AirPlaneDesign 0.4.1
  * btl 0.9.9
  * CurvedShapes 1.0.13
  * Curves 0.6.61
  * fasteners 0.5.38
  * PieMenu 1.11.3
  * Road 2025.2.22
  * sheetmetal 0.7.22
  * Telemetry 1.0.0

Katt-Kushol, 2025-05-03:
Boolean cut was success in FC1.1dev version. 

[image]

Version Info:
`OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41365 (Git)
Build date: 2025/04/22 11:42:50
Build type: Release
Branch: main
Hash: 84b57d67b50ac64be2246418fb4b49406904f3a9
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/87.9548/1
Installed mods: 
  * CurvedShapes 1.0.13
  * Curves 0.6.60
  * DynamicData 2.71.0
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * lattice2 1.0.0
  * Road 2025.2.6
  * sheetmetal 0.7.22
`

Pendelf, 2025-05-03:
Indeed in Part works fine, in PartDesign, only if the thread is turned off 

https://github.com/user-attachments/assets/cefec4bb-fdc1-41e2-a016-e86877507a9a

maxwxyz, 2025-05-04:
@pieterhijma might this be an OCC issue?

FlachyJoe, 2025-05-04:
It works with the original thread object but not with a **refined** SubShapeBinder

[image]

This refine feature is a disaster, and its auto-appliance is worse !
```
OS: Debian GNU/Linux trixie/sid (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.41491 (Git)
Build date: 2025/05/02 22:58:22
Build type: Unknown
Branch: main
Hash: d427162d976b5724c4bd9a985fc16fb898dc4697
Python 3.13.3, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.16
shiboken 5.15.16, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.8.1
Locale: English/United States (en_US) [ OS: English/United States (en_US) ]
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.4394/1
Installed mods: 
  * Render 2024.12.15
  * lattice2 1.0.0
  * Lithophane
  * Curves 0.6.61
  * fasteners 0.5.38
  * Assembly4 0.50.18
  * Telemetry 1.0.0
  * Road 2025.2.22
  * FCPDWorkbench 0.4.0
  * sheetmetal 0.7.22

```

luzpaz, 2025-05-05:
Can someone suggest a rename of the ticket so it's more accurate ?

pieterhijma, 2025-05-14:
> [@pieterhijma](https://github.com/pieterhijma) might this be an OCC issue?

May still be, but since the boolean cut works in Part, I believe it is more likely to be an issue with FreeCAD.

FlachyJoe, 2025-05-14:
@pieterhijma the issue is multiple:

1. In some cases the Refine method doesn't create good shape. I don't know if it's an OCC problem or the way we call OCC functions. Related #17846
2. When the Refine produces erroneous shape the user is not informed about the source of the problem.
3. We can't switch Refine off at feature creation. So when the Auto-Refine preference is set some feature can't be created at all #18110

pieterhijma, 2025-05-14:
Ok, thanks for letting me know, but the fact that there is a route to correct behavior (via Part booleans) does not give me enough evidence to investigate this further to find out whether it is OCC related. 

However, if it is indeed the case that the refine method doesn't create a good shape, then this may be OCC related. Could you perhaps provide me with a clear example where the (auto) refine doesn't create a proper shape and create a new issue for that (or direct me to one you already know). If you ping me on that issue, I'll investigate that then.

FlachyJoe, 2025-05-14:
@pieterhijma with the file of this issue:

1. Create a SubShapeBinder of the thread (its Refine property is True even if you have disable the autorefine preference).
2. Do a PartDesign boolean. It fails.
3. Switch off the SubShapeBinder Refine property: the boolean works.

#17846 has many other examples.

FlachyJoe, 2025-05-17:
FYI the PartDesign boolean cut works with

> OS: Debian GNU/Linux trixie/sid (XFCE/xfce/xcb)
> Architecture: x86_64
> Version: 1.1.0dev.41726 (Git)
> Build date: 2025/05/16 17:15:43
> Build type: Unknown
> Branch: main
> Hash: 1ca108f8d503b2b68455844bfdfabb62fa1461ed
> Python 3.13.3, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.16
> shiboken 5.15.16, SMESH 7.7.1.0, xerces-c 3.2.4, **OCC 8.0.0.rc1**-7aa85582ad
> Locale: English/United States (en_US) [ OS: French/France (fr_FR) ]
> Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
> Logical DPI/Physical DPI/Pixel Ratio: 85/86.4394/1
> Installed mods: 
>   * Render 2024.12.15
>   * Curves.backup1744720429.3214982 0.6.60 (Disabled)
>   * lattice2 1.0.0
>   * Lithophane
>   * Curves 0.6.62
>   * fasteners 0.5.38
>   * Assembly4 0.50.18
>   * Telemetry 1.0.1
>   * FreeCAD_SketchArch
>   * Road 2025.2.22
>   * FCPDWorkbench 0.4.0
>   * sheetmetal 0.7.22

[image]

luzpaz, 2025-11-28:
> https://github.com/FreeCAD/FreeCAD/issues/21097#issuecomment-2879853895

@FlachyJoe this a duplicate then ?

FlachyJoe, 2025-11-29:
@luzpaz duplicate of #17846 which can be re-title "PartDesign: Automatic refine cause issues with Helixes and threaded shapes"

closed 2025-11-29 by maxwxyz