FreeCAD/FreeCAD #21105: BOLTS Parts Selector List is Too Tiny

KitchM, 2025-05-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

After starting the start_bolts.FCMacro, there is a right-side panel that displays a list of parts.  However, it is so tiny that it is hard to navigate and to view.  (See cursor arrow in screenshot.)

This should be larger and/or expandable.

Thank you.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * boltsfc 2022.11.5
  * Manipulator 1.5.9
  * dodo 1.0.1
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-03:
Please open an issue in the BOLTS repo: https://github.com/boltsparts/BOLTSFC

closed 2025-05-03 by maxwxyz

KitchM, 2025-05-04:
Okay, will do. Thanks.

KitchM, 2025-05-04:
Well, I gave it a shot.  However, there is no Issue option at that web site, unless I missed something.  Please see attached 

[image]

screenshot.